riscv-software-src/riscv-isa-sim #14: Dynamic instruction count

channoh, 2015-03-23:
Hi,

I want to get dynamic instruction count via Spike, but the disassembly codes from -d option (i.e., ```spike -d pk hello``` and ```:r```) is weird.

aswaterman, 2015-03-23:
The proxy kernel can do this for you with its -s flag (try spike pk -s
hello)

channoh, 2015-03-23:
Good. Thanks!

closed 2015-03-23 by channoh

AhmedSamara, 2018-06-09:
Is there an updated and current way to get dynamic instruction count? This no longer works.

AhmedSamara, 2018-06-20:
Is there a way to get a more detailed breakdown of types of instructions executed?   
And if not, would there be an interest in me implementing this feature and upstreaming it?

anmolsahoo25, 2018-06-20:
Hi,
We are working on a Python tool which takes as input the log file generated from spike using the -l option and outputs different statistics based on the log file.

You can check out the implemented features and ongoing features at - https://bitbucket.org/anmol07/spikepost

AhmedSamara, 2018-06-20:
Can that one differentiate between instructions in the user program vs instructions from pk?

anmolsahoo25, 2018-06-21:
Currently we are processing log files using pure Python. I am looking at a numpy/spark backend for faster analytics. As to the other part of your question, I am not sure if I understand correctly. Could you please elaborate a bit about what you mean instructions generated by pk?

AhmedSamara, 2018-06-21:
What I mean is when you execute a program with `spike pk <prog>`, you aren't just running your program, you are also running pk. So not every instruction that is a part of the log is a part of the program. 
But as I type this out I realized pk is only necessary for me because of the printf statements.

neelgala, 2018-06-21:
You are right. Syscalls like printf will be inserted by pk. In
spikepost/Rita we also plan to add a feature of skip-address.. where the
profiling skips sections of the user specified address range for all or
certain histograms.

Again this is work in progress. Might it take its time for the feature to
get plugged in.

anmolsahoo25, 2018-06-21:
We already have a feature of selecting an address range where you want the profiling to happen. It can be extended to incorporate address blocks that have to be skipped. Will add that as an enhancement in the repo.

AhmedSamara, 2018-07-29:
I think the problem with using a full log file is that for complicated programs, it becomes really huge. for the program I was trying to benchmark, it produced a 300GB log file and then crashed because it run out of HDD space. I could move it onto a larger system but at that point, parsing will also be ridiculous. It's possible I did something wrong.
I'm working on an instruction count modification on my fork and will submit a PR if I can get anything reasonable.