riscv-software-src/riscv-isa-sim #36: Command line processing issue?

sbates130272, 2016-01-05:
Hi

I am seeing an issue with processing the -p and -m options for spike? Note that I pulled the repo as a submodule of the riscv-tools repo.

Stephen

root@c6403e37998e:/opt/riscv# spike -m 128 +disk=linux-3.14.41/root.bin bbl linux-3.14.41/vmlinux quit
spike: argument required for option -m
usage: spike [host options] <target program> [target options]
Host Options:
  -p <n>             Simulate <n> processors [default 1]
  -m <n>             Provide <n> MiB of target memory [default 4096]
  -d                 Interactive debug mode
  -g                 Track histogram of PCs
  -l                 Generate a log of execution
  -h                 Print this help message
  --isa=<name>       RISC-V ISA string [default RV64IMAFDC]
  --ic=<S>:<W>:<B>   Instantiate a cache model with S sets,
  --dc=<S>:<W>:<B>     W ways, and B-byte blocks (with S and
  --l2=<S>:<W>:<B>     B both powers of 2).
  --extension=<name> Specify RoCC Extension
  --extlib=<name>    Shared library to load

sbeamer, 2016-01-05:
Oddly enough, the numerical command line arguments should have no space between the number and the flag letter, e.g.:
spike -m128 +disk=linux-3.14.41/root.bin bbl linux-3.14.41/vmlinux

aswaterman, 2016-01-05:
This is a bug in the help message (there should be no spaces)--I've fixed
it.  Thanks.

closed 2016-01-05 by aswaterman

sbates130272, 2016-01-05:
Thanks! (though it would be nice if the code could be updated to handle a space) ;-).