riscv-software-src/riscv-isa-sim #38: Loads of "Multiple definition of.." errors when building - since last commit

DerManiac, 2016-03-02:
The last commit [a95b44df9d142a49f3e1bfe4a60d259ed41ca247](https://github.com/riscv/riscv-isa-sim/commit/a95b44df9d142a49f3e1bfe4a60d259ed41ca247#diff-5dde698536db884df34d19421322723a) seemed to have broken the simulator for me. Now when building riscv-isa-sim (or the entire riscv-tools) I get a huge amount of errors that all go something like this:

`ui64_to_f64.o: In function `softfloat_shortShiftRightJam128':
ui64_to_f64.c:(.text+0x180): multiple definition of `softfloat_shortShiftRightJam128'
f32_add.o:f32_add.c:(.text+0x180): first defined here
ui64_to_f64.o: In function `softfloat_shortShiftRightJam128Extra':
ui64_to_f64.c:(.text+0x1c0): multiple definition of `softfloat_shortShiftRightJam128Extra'
f32_add.o:f32_add.c:(.text+0x1c0): first defined here
ui64_to_f64.o: In function `softfloat_shiftRightJam64Extra':
ui64_to_f64.c:(.text+0x210): multiple definition of `softfloat_shiftRightJam64Extra'
f32_add.o:f32_add.c:(.text+0x210): first defined here
`

When reverting to `bea283531abfd013811e83c75ff6189a0d9b3075` it builds fine.

aswaterman, 2016-03-02:
Thanks for letting us know. Unsurprisingly, we did not observe this :-)
What OS and compiler are you using?

DerManiac, 2016-03-02:
I'm using Fedora 23. Here is the output of gcc -v:

    Using built-in specs.
    COLLECT_GCC=gcc
    COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/5.3.1/lto-wrapper
    Target: x86_64-redhat-linux
    Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,objc,obj-c++,fortran,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --disable-libgcj --with-isl --enable-libmpx --enable-gnu-indirect-function --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
    Thread model: posix
    gcc version 5.3.1 20151207 (Red Hat 5.3.1-2) (GCC)

colinschmidt, 2016-03-02:
Just another data point, I have the same issue with

```
Using built-in specs.
COLLECT_GCC=gcc-5
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.3.0-3ubuntu1~14.04' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=gcc4-compatible --disable-libstdcxx-dual-abi --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.3.0 20151204 (Ubuntu 5.3.0-3ubuntu1~14.04
```

but not

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.9/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.9.3-5ubuntu1~14.04' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.9.3 (Ubuntu 4.9.3-5ubuntu1~14.04)
```

aswaterman, 2016-03-02:
Does this patch fix it?
    diff --git a/softfloat/platform.h b/softfloat/platform.h
    index 3d9c431..03dd429 100644
    --- a/softfloat/platform.h
    +++ b/softfloat/platform.h
    @@ -44,5 +44,5 @@ SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
     
     /*----------------------------------------------------------------------------
     *----------------------------------------------------------------------------*/
    -#define INLINE extern inline
    +#define INLINE static inline

aswaterman, 2016-03-02:
That didn't paste very well, but I think it gets the point across (extern inline -> static inline in softfloat/platform.h)

DerManiac, 2016-03-02:
Yup, that seems to have done it! Thanks!

closed 2016-03-02 by DerManiac

aswaterman, 2016-03-02:
Pushed to master; thanks again.