riscv-software-src/riscv-isa-sim #41: master fails build because it can't make devicetree.h

timsifive, 2016-05-19:
tnewsome@debian-lap:~/SiFive/riscv-tools/riscv-isa-sim$ make
make: *** No rule to make target 'riscv/devicetree.h', needed by 'all'.  Stop.

Is this a file you forgot to commit?

acw1251, 2016-05-19:
I guess this came from my recent pull request. It was a bug fix for the version of spike in riscv-tools, but I accidentally put the target of the pull request as master. devicetree.h used to exist, but I guess it has been deleted since.

Anyways, what's the correct way to send a pull request for the riscv repos in riscv-tools?

sbeamer, 2016-05-19:
Submit a pull request to the affected repo. If you only want to change the submodule pointer in riscv-tools, you can submit a pull request to riscv-tools.

timsifive, 2016-05-19:
Thanks for fixing this.

closed 2016-05-19 by timsifive