riscv-software-src/riscv-isa-sim #45: Loading flat, non-ELF files

neuschaefer, 2016-05-25:
To test Boot ROM images (at least to some limited degree), it would be useful to be able to load files at some common positions in the address space (e.g. at the start), without first having to build an ELF file.

Are there any plans to implement such a feature?

aswaterman, 2016-05-25:
Yep.  Probably something that can be fed both to Spike and to our hardware implementations (to be baked into ROM at synthesis time, in that case).  But we're gonna hold off until after debug is fully implemented and the current HTIF is kaput.

aswaterman, 2016-05-25:
That said, ELF is a pretty reasonable file format for this task.

maaku, 2016-06-11:
My organization has a direct application for this feature -- we have a codebase that needs to be kept to minimal size, and the code to build and parse ELF headers is both unnecessary and a potential security hazard to us.

timsifive, 2016-06-14:
I just pushed changes that make gdb work with 32-bit spike at
https://github.com/riscv/riscv-isa-sim/tree/debug so now both 32- and
64-bit are supported.

You can use this with gdb's restore command to load code straight from eg.
ihex or srec files. If you try it, let me know how it goes.

Tim