riscv-software-src/riscv-isa-sim #50: Autotools: pre-compile C headers with CC, not CXX

maaku, 2016-06-06:
Compiling on Mac OS X 10.11 results in the following error:

    g++ -fPIC -MMD -MP  -Wall -Wno-unused -g -O2 -std=c++11 -DPREFIX=\"/usr/local\" -Werror -I. -I. -I./riscv -I./dummy_rocc -I./softfloat -I./spike_main ./riscv/insn_template.h -o insn_template.h.gch
    clang: error: treating 'c-header' input as 'c++-header' when in C++ mode, this behavior is deprecated
    make: *** [insn_template.h.gch] Error 1

This is because a C header is being pre-compiled with the C++ compiler, something Clang does not support.

maaku, 2016-06-14:
Interestingly this isn't a problem when compiling the riscv-tools repo. `-Werror` is the likely culprit. I haven't dug in to see why it is set by default when compiling riscv-isa-sim, but not when compiling via riscv-tools.

timsifive, 2016-06-14:
I set -Werror explicitly in the riscv-isa-sim Makefile a while back,
reasoning that the code was compiling cleanly and this is a good way to
keep it that way.
How much work is required to make the code clang clean as well?

Tim