riscv-software-src/riscv-isa-sim #63: CSR_*TIME not implemented

neuschaefer, 2016-08-23:
I noticed that, while `CSR_TIME` and `CSR_STIME` are [redirected](https://github.com/riscv/riscv-isa-sim/blob/master/riscv/processor.cc#L425) to `CSR_MTIME`, `CSR_MTIME` is [not implemented](https://github.com/riscv/riscv-isa-sim/blob/master/riscv/processor.cc#L445), resulting in an "Illegal Instruction" exception everytime one of these CSRs is used. Although bbl seems to work around the lack of `mtime` somehow, it would be nice to have support for it in spike.

aswaterman, 2016-08-23:
As with mtimecmp, we are leaning towards requiring the platform provide
this register through its preferred means, rather than mandating it be a
CSR.

The rationale is that the RTC necessarily lives in a different clock/power
domain, and so is not part of the hart (hence CSR).

neuschaefer, 2016-08-23:
I guess the privileged spec 1.9 is outdated in this regard; it specifies: "M-mode includes a timer facility provided by the high-resolution read-only real-time counter mtime.", which does not sound optional.

Does spike use a memory-mapped RTC instead of mtime?

aswaterman, 2016-08-23:
Yeah, we have rethought that decision. Sorry for being a pain in the ass :-/

colinschmidt, 2016-09-25:
I just ran into this issue trying to get this commit of [riscv-linux](https://github.com/riscv/riscv-linux/commit/8205b66a1104171284699985409ddd4d6921400d) to boot. 
How should rdtime resolve itself in spike? 

Its unclear to me whether the csr_time case should be re-added to the get_csr switch statment, or if the rdtime psuedo op needs to be updated to not use a csr? or something else entirely? for now I can pretty easily work around it but I was thinking about submitting a PR but I wasn't sure what the right solution was.