riscv-software-src/riscv-isa-sim #69: Cleaning up Spike for clarity and flexibility over performance

acw1251, 2016-09-20:
Since Spike is the main reference implementation of RISC-V, I think it should be flexible and easy to understand. Since we have qemu support for faster simulation, I think we can afford some loss of performance in favor of configurability and cleanliness.

Here are some of the things I had in mind:
* Separate hardware that is not part of the processor into a platform_t class. This includes dram, rtc, debug module. This could easily be extended to include a custom uart for processors that may not have a debug module.
    * One possible constructor for this class could take in a configuration string as an input and parse it to find what devices exist and where they go.
* Move the "icache" entirely within processor_t to get instruction decoding out of the mmu. Also rename it to something else since its really caching decodings and trying to providing unique call sites for each decoded instruction function to help the host next address predictor (maybe "execution cache"?).
* Have tlb store physical addresses instead of pointers to host memory offset by the full virtual address
* Remove 
* Set processor interrupts through methods, not through changing the state directly.
* Get rid of as many friend classes as possible and get classes to communicate through well defined interfaces that can be reimplemented for various purposes (e.g. new mmu_t class for weaker memory models).
*

These are seom
* Support different platforms through inputting a configstring
* Support different implementation details through configuration
* Support further configuration

It should be flexible enough to support different platform configurations (i.e. where dram starts, if you have a debugger or not) and different implementation details (i.e. which misaligned loads and stores succeed and which trigger exceptions). It should be cleanly written so there are clear divisions between modules 

I started working on adding more configurability to Spike, and I noticed that it's use of friend classes and constants makes it harder to modify.

Since we have qemu support for fast simulation, I think it would be okay for Spike to lose some performance to gain flexiblitiy and cleanliness.

Since we have qemu support for simulating RISC-V programs very fast, do we need optimizations in 

Right now there are three simulators in the official riscv github group (spike, qemu, and angel). Qemu

My understanding is 

Since qemu is much faster than spike for simulation

acw1251, 2016-09-20:
Here is my fork of spike: https://github.com/acw1251/riscv-isa-sim
I've done some of the things mentioned in the list above, but there is still a lot of work to do to get the platform class I was proposing.

I also have a simple but messy flex + bison parser for configuration strings here: https://github.com/acw1251/riscv-configstring-parser

In case you are going off of e-mail, I submitted the issue before I finished, and I have since edited and finished the issue.

evancox10, 2017-10-13:
I looked at your code Andy, this is good work. I'm trying to understand spike right now, and IMO the current code base makes this kind of hard. I don't think the situation has improved greatly over the last year.