riscv-software-src/riscv-isa-sim #72: reading registers from gdb kills spike --isa=RV32IM

vanjoe, 2016-10-06:
When I run `spike --isa=RV32IM -H --gdb-port=7777`  and after connecting with gdb I run `info 
registers`

Spike dies with the the error message 

`gdbserver code died: handle_register_read`

gdb displays the following

```
ra             0x0000000000000000	0
sp             0x8000000000000000	-9223372036854775808
gp             0x0000000000000000	0
tp             0x0000000000000000	0
t0             0x0000000000000000	0
t1             0x0000000000000000	0
t2             0x0000000000000000	0
fp             0x0000000000000000	0
s1             0x0000000000000000	0
a0             0x0000000000000000	0
a1             0x0000000000000000	0
a2             0x0000000000000000	0
a3             0x0000000000000000	0
a4             0x0000000000000000	0
a5             0x0000000000000000	0
Remote connection closed
```

I think gdb is trying to read 64* 32 bits of registers rather than 32*32 bits like it should.

timsifive, 2016-10-06:
I'll take a look.

timsifive, 2016-10-07:
As a workaround, you can tell gdb: `set arch riscv:rv32`

This may be necessary anyway, since I don't think there's a way for gdb to discover the architecture of the target using the gdb protocol. (I'd love to be proven wrong.) At least though I'm going to make spike not crash.

vanjoe, 2016-10-07:
Could gdb ask for the misa csr and change if it's implemented and reports
rv32?

timsifive, 2016-10-07:
> Could gdb ask for the misa csr and change if it's implemented and reports rv32?

That's a good idea. I'll take a look, although I'm not hopeful that it'll be easy to implement.