riscv-software-src/riscv-isa-sim #74: Simulator fails with 'trap_store_access_fault' with any test

alexmechanic, 2016-10-19:
Any test or benchmark fails at start with error `trap_store_access_fault`.
any-bit pk.
Why spike can deny storing io?

evancox10, 2017-10-21:
This happens if you try to load the ELF file to somewhere other than RAM. elf loading goes through HTIF and writes to the sim's bus device. There must be a writable device here.

...At least that's the behavior a now, a year and a day after you posted this. I thought I'd post the explanation so other people could find the reason why this happens.

Make sure that the elf _entry symbol has a RAM big enough to hold the code at that location.

alexmechanic, 2017-10-21:
@evancox10 I still have this as my project, it's just conservated for now.
I'm sure I can use that info later, thank you

alexmechanic, 2018-10-23:
Sorry, currently I left working with this project and cannot force this issue anymore.

closed 2018-10-23 by alexmechanic