riscv-software-src/riscv-isa-sim #79: Debugging With Gdb not work

colin4124, 2016-12-15:
I use riscv tools version: 886d8131dbd23533fb04d2d76a80be21d5f9ee7a and riscv-isa-sim version: 53d74f4cc31ecf5bb6499b886d8fcbc992a17920

First I compile `hello.c` with `-g` to `hello.rv`:
```
$ riscv64-unknown-elf-gcc -g hello.c -o hello.rv
```

Then I follow README https://github.com/riscv/riscv-isa-sim#debugging-with-gdb  in one shell run:
```
$ spike --gdb-port 9824 pk hello.rv
```
It quickly prints "hello, word!"

So when I in a second shell I do that:
```
$riscv64-unknown-elf-gdb hello.rv
(gdb) target remote localhost:9824
```
It returns "localhost:9824: Connection timed out."

By the way, can I use rocket-chip/emulator with riscv-gdb like the way spike do so?

timsifive, 2016-12-16:
Does your program have a while() loop as in the example? If not, it will run to completion before you get to attach with gdb (as you discovered).
If you don't want to add the loop, you can run spike with -H, which causes it to start in debug mode. Note that this is halted before the kernel is even started. There's a lot of code that has to run before the actual program you want to execute gets to run. Using the loop is much simpler.

colin4124, 2016-12-19:
@timsifive Thinks very much!

 [riscv-tests/debug](https://github.com/riscv/riscv-tests/tree/master/debug) demonstrates how to debug on these target (spike32/64, fe300 and fu500). I would take a look at it and learn how to debug.

By the way, there isn't C simulator target and VCS simulator target. Could I debug C simulator target (emulator-rocketchip-DefaultConfig) with openOCD( JTAG ) as target fe300/fu500 do so?

timsifive, 2016-12-19:
In theory rocket chip has JTAG and a debug module in it, so if you can figure out how to have OpenOCD talk to it that should be possible. I know some people have had OpenOCD talk to a Verilog simulator using jtag_vpi. It's very slow, though.

colin4124, 2016-12-20:
@timsifive Thinks a lot!

closed 2016-12-20 by colin4124