riscv-software-src/riscv-isa-sim #84: Debug exceptions get overwritten by S1

bacam, 2016-12-21:
The debug ROM writes 1s to the last word in the debug RAM to indicate an exception.  However, it also uses the last word for stashing the value of S1 before running the code in the debug RAM, so the exception information is lost.  The gdb support checks for exceptions in a couple of places and can misbehave when S1 is non-zero.  In particular, it will send an error code for `p` and `P` commands, although the latter is masked by an extra `OK` response (see pull request #83).

timsifive, 2016-12-29:
I don't quite follow this. What should happen is:
1. Before jumping to Debug RAM, store s1 to the last word in Debug RAM.
2. When re-entering Debug RAM, restore s1 from the last word in Debug RAM.
3. Then write either 0 or ~0 to the last word in Debug RAM, depending on whether there was an exception or not.

Here's what it looks like when an exception happens:
```
core   0: 0x0000000000000400 (0x41002403) lw      s0, 1040(zero)
core   0: 0x0000000000000404 (0x41400483) lb      s1, 1044(zero)
core   0: 0x0000000000000408 (0x00940023) sb      s1, 0(s0)
core   0: exception trap_store_access_fault, epc 0x0000000000000408
core   0:           badaddr 0xffffffffdeadbeef
core   0: 0x0000000000000808 (0xfff00413) li      s0, -1
core   0: 0x000000000000080c (0x0080006f) j       pc + 0x8
core   0: 0x0000000000000814 (0x0ff0000f) fence
core   0: 0x0000000000000818 (0x301024f3) csrr    s1, misa
core   0: 0x000000000000081c (0x0004c663) bltz    s1, pc + 12
core   0: 0x0000000000000820 (0x43c02483) lw      s1, 1084(zero)
core   0: 0x0000000000000824 (0x0080006f) j       pc + 0x8
core   0: 0x000000000000082c (0x42802e23) sw      s0, 1084(zero)
```

And here when there are no problems:
```
core   0: 0x0000000000000400 (0x41f02823) sw      t6, 1040(zero)
core   0: 0x0000000000000404 (0x4000006f) j       pc + 0x400
core   0: 0x0000000000000804 (0x00c0006f) j       pc + 0xc
core   0: 0x0000000000000810 (0x00000413) li      s0, 0
core   0: 0x0000000000000814 (0x0ff0000f) fence
core   0: 0x0000000000000818 (0x301024f3) csrr    s1, misa
core   0: 0x000000000000081c (0x0004c663) bltz    s1, pc + 12
core   0: 0x0000000000000820 (0x43c02483) lw      s1, 1084(zero)
core   0: 0x0000000000000824 (0x0080006f) j       pc + 0x8
core   0: 0x000000000000082c (0x42802e23) sw      s0, 1084(zero)
```

In this case 1084(zero) is the last word in Debug RAM.

bacam, 2016-12-30:
Sorry, I somehow got the idea into my head that the check for an exception was done through instructions in debug RAM, but the gdb code does it directly, of course.

However, setting the value of S1 does cause it to falsely detect exceptions.  For example, if I start spike with execution halted, connect with gdb and do `set $s1 = 10` the debugging code logs the following:

    Received 23-byte packet from debug client: $P9=0a00000000000000#f7
    wrote 7 bytes: +$OK#9a
    read32(15) -> 0x0
    read32(14) -> 0xa
    wrote 7 bytes: $E03#a8

(The first response is the duplicate `OK` fixed in #83.)  The problem that causes the `E03` error is that it doesn't just read the last 32 bits to check for an exception, but the last 64 bits.  This still contains half of S1.  I've made a patch for this and I'll send a pull request in a moment.

timsifive, 2016-12-30:
What a fun bug. Thank you for sticking with it and tracking this down.

PR #85 (2016-12-30): Only read exception flag in gdb register read/write.
The flag is 32 bits, and if we read 64/128 bits then we get fragments of
S1 too and can accidentally send an error.  Fixes #84.
commit f38dcde0 (2016-12-30): Only read exception flag in gdb register read/write. (#85)

The flag is 32 bits, and if we read 64/128 bits then we get fragments of
S1 too and can accidentally send an error.  Fixes #84.
commit 67c1228b (2017-01-17): Only read exception flag in gdb register read/write. (#85)

The flag is 32 bits, and if we read 64/128 bits then we get fragments of
S1 too and can accidentally send an error.  Fixes #84.
closed 2016-12-30 by aswaterman