riscv-software-src/riscv-isa-sim #88: Trap Return Instructions

zhanghongce, 2017-01-07:
Hi,

I wonder if there is a mismatch between the privilege-level spec and the Spike simulator.

In the spec (Version 1.9.1 released on 11/04/16) page 22, the semantics of xRET are described as follows,

The MRET, HRET, SRET, or URET instructions are used to return from traps in M-mode, H-mode,  S-mode,  or  U-mode  respectively.   When  executing  an xRET  instruction,  supposing xPP holds the value y, yIE is set to xPIE; the privilege mode is changed to y; xPIE is set to 1;  and xPP is set to U.

Which means if we execute MRET in M-mode, and returns to say S-mode, MPIE should be set to 1, but in the simulator, (riscv/insns/mret.h)

s = set_field(s, MSTATUS_MPIE, 0);

Could anyone tell if this is indeed a mismatch or just my misunderstanding?

Thank you!
Hongce

aswaterman, 2017-01-07:
Yeah, this is benign.  Setting xPIE=1 is meant to detect software bugs; it
should not matter what happens to the xPIE field after a yRET.  You're
welcome to submit a pull request.

zhanghongce, 2017-01-08:
Thank you for your reply! But I didn't get it why setting xPIE=1 can be used to detect software bugs.  So can we treat the spec as saying xPIE can be anything after executing xRET?

sorear, 2017-01-08:
If a SRET is followed by a non-delegated trap into M-mode, the M-mode code can observe the value of SPIE which was written by SRET.

aswaterman, 2017-01-09:
By "benign" I meant it doesn't create a security hole. It's still a bug and Spike needs to be fixed. I'll do so shortly.

zhanghongce, 2017-01-09:
Thank you very much for your reply!
Sorry for bothering you again, but I just have one more question. It seems legal to execute an SRET instruction when in M-mode. Is this behavior desired?

aswaterman, 2017-01-09:
It is legal.  In practice, it is not likely to be especially useful, but allowing all instructions in M-mode simplifies testing.

closed 2017-02-02 by aswaterman

mainkar-rohit, 2019-02-19:
Hi Andrew,
It is legal to execute sret in M mode only if S mode is available right? 
If S mode is not supported, isn't sret illegal?

aswaterman, 2019-02-19:
That’s right.

mainkar-rohit, 2019-02-19:
Thanks for clarifying. Is Spike always expected to have S & U mode?
It looks like currently these corresponding bits in misa are hardcoded to 1.

aswaterman, 2019-02-19:
Spike doesn't support configuring which privilege modes are available (yet).

commit d392463e (2017-02-02): Set xPIE=1 on xRET

Resolves #88.