riscv-software-src/riscv-isa-sim #93: The unaligned load/store is not supported in Spike

zhenbohu, 2017-04-06:
Hi,

When I use the spike as the golden model to run a tests, I found the spike is not supporting the unaligned load/store instructions. But in ISA, this is the baseline feature which have to be supported. Is there any reason why Spike not support it, any plan to fix this issue?

Thanks
Bob

terpstra, 2017-04-06:
The spec only requires that the combination of hardware + firmware support
it. The riscv-pk/bbl firmware / boot loader handles them.

newtonchen, 2017-04-06:
As a isa simulator, I think spike should support unalgined load/store. ISA does not forbid it to be implemented solely by HW.

terpstra, 2017-04-06:
It also does not require it. The way it is now, it can simulate the trap
handling, which I would argue is more useful.

newtonchen, 2017-04-06:
So if a HW implementation can support this feature, how can it use spike for trace checking? I think spike should serve as a golden model in HW verification and support all possible features allowed in ISA. Work around in verification flow is error prone .

terpstra, 2017-04-06:
If you want this feature, submit a patch for review which makes spike
optionally support evaluating misaligned load/stores.

aswaterman, 2017-04-06:
I agree with @terpstra that it's more useful to model the ISA-level behavior of the most common hardware implementations as closely as possible.  Among other things, that lets us test the firmware.

Nevertheless, I'm in the progress of adding this as a configure-time option...

sorear, 2017-04-06:
If you want a model that will execute exactly the same trace as a given HW implementation, then that model needs to match the implementation choices made in that specific implementation.

In particular Rocket does *not* do HW unaligned accesses, so if your firmware targets Rocket you need an unaligned access trap handler, which needs to be tested somehow.

BTW QEMU does do misaligned accesses natively, but it doesn't generate instruction traces so it may not be useful to you.

aswaterman, 2017-04-06:
This is implemented in 1132fdf (but not yet merged into master)

closed 2017-04-06 by aswaterman

newtonchen, 2017-04-06:
@aswaterman Thanks, we probably will defer unaligned testing and wait for spike support.

@sorear Thanks, guess we need to disable HW support for now, as some of the asm tests in riscv-tests also assume no unalgined support.

sorear, 2017-04-06:
@newtonchen That would (IMO) be a bug in riscv-tests.  HW implementations of unaligned access are definitely supposed to be allowed, riscv-tests should check for either (misaligned-access trap generated OR access done directly with the correct behavior)

aswaterman, 2017-04-06:
I believe only one test is getting that wrong.  I'll see if I can quickly fix it.

newtonchen, 2017-04-06:
@aswaterman I think it's the riscv-tests/isa/rv64mi/ma_addr.S, thanks.

aswaterman, 2017-04-06:
https://github.com/riscv/riscv-tests/commit/ce4b19d7b75cd024b81c09fb98c99bf4298ae7ad

commit 1132fdf4 (2017-04-06): Add --enable-misaligned option for misaligned ld/st support

Resolves #93