riscv-software-src/riscv-isa-sim #98: Segmentation fault when simulating bbl

poinu, 2017-05-15:
Hello!
We are getting a segmentation fault when trying to simulate bbl in recent versions of this repository.

In particular, we are using commit `75f2a05df9cdff6f3faba748065b3184b9f01b01` and, after some debugging, we could isolate the segmentation fault in the *line 174* in file `../riscv/mmu.h`.

The corresponding code is as follows:

```
  inline icache_entry_t* refill_icache(reg_t addr, icache_entry_t* entry)
  {
    auto tlb_entry = translate_insn_addr(addr);
    insn_bits_t insn = *(uint16_t*)(tlb_entry.host_offset + addr); // <---- This is line 174!
    int length = insn_length(insn);

    if (likely(length == 4)) {
      insn |= (insn_bits_t)*(const int16_t*)translate_insn_addr_to_host(addr + 2) << 16;
    } else if (length == 2) {
      insn = (int16_t)insn;
    } else if (length == 6) {
      insn |= (insn_bits_t)*(const int16_t*)translate_insn_addr_to_host(addr + 4) << 32;
      insn |= (insn_bits_t)*(const uint16_t*)translate_insn_addr_to_host(addr + 2) << 16;
    } else {
      static_assert(sizeof(insn_bits_t) == 8, "insn_bits_t must be uint64_t");
      insn |= (insn_bits_t)*(const int16_t*)translate_insn_addr_to_host(addr + 6) << 48;
      insn |= (insn_bits_t)*(const uint16_t*)translate_insn_addr_to_host(addr + 4) << 32;
      insn |= (insn_bits_t)*(const uint16_t*)translate_insn_addr_to_host(addr + 2) << 16;
    }

    insn_fetch_t fetch = {proc->decode_insn(insn), insn};
    entry->tag = addr;
    entry->data = fetch;

    reg_t paddr = tlb_entry.target_offset + addr;;
    if (tracer.interested_in_range(paddr, paddr + 1, FETCH)) {
      entry->tag = -1;
      tracer.trace(paddr, length, FETCH);
    }
    return entry;
  }
```
We have also tried commit `63c98b41ae2d36808ebbb57e7f8e871ba247d444`, where the segfault doesn't occur anymore. However, we could see how spike was stuck in that code region ignoring the previous segfault.

aswaterman, 2017-05-15:
Since the segfault is fixed, I'm not sure what to do here.  bbl works correctly for me.

poinu, 2017-05-16:
We are currently on commit `e465de145c69c28230b5c73cc58f96fd9bd04419` while using commit `cf7369999779bc78bcc5a8a6fe6187363b23247a` of the RISCV tools and have tried  running thebbl both with and without payload.

When running the bbl **without** payload, everything works as expected and we get the following output:
```
This is bbl's dummy_payload.  To boot a real kernel, reconfigure
bbl with the flag --with-payload=PATH, then rebuild bbl.
```

When running the bbl **with** a Linux kernel as a payload, `spike` seems to hang (see attached spike trace file) where it used to segfault in the previous commit `75f2a05df9cdff6f3faba748065b3184b9f01b01`. We are actually using commit `9079be6167aec516a4625b7dcf20efbf384277b4` of the [riscv-linux](https://github.com/riscv/riscv-linux) repository.

In case it is useful, the configuration line used to generate such bbl is the following:
```
../configure --prefix=path/to/riscv-tools/installation --host=riscv64-unknown-elf host_alias=riscv64-unknown-elf RISCV=path/to/riscv-tools/installation --with-payload=path/to/linux/vmlinux --disable-logo
```

If simulating a bbl with payload is working for you, would you mind telling me which precise commits of the repositories are you using?

In file [spike_execution_trace.txt](https://github.com/riscv/riscv-isa-sim/files/1005307/spike_execution_trace.txt) you can find a trace where a "full iteration" of spike is shown.

aswaterman, 2017-05-16:
Ah, so, it's not Spike that's hanging; the kernel or BBL is hanging.  The question is, what causes the target software to jump to address 0?

FYI, I'm using the HEAD of the riscv-next branch of riscv-linux; HEAD of priv-1.10 of riscv-pk; and HEAD of priv-1.10 of riscv-isa-sim.  That combination seems to work.

poinu, 2017-05-20:
You are absolutely right, coming from an older build I was using the wrong branch of the riscv-linux repository. Thank you for your time. for pointing me to the right branches, and sorry for the confusion!

closed 2017-05-20 by poinu