riscv-software-src/riscv-isa-sim #102: gdb-port option unrecognized

TheThirdOne, 2017-05-19:
The readme shows  `spike --gdb-port 9824 pk tests/debug` as a way to set up debugging with gdb, but running this command doesn't work as the gdb-port option has been replaced with the rbb-port option as of  d1f2cf337. 

```
$ spike --gdb-port 9824 pk sample
spike: unrecognized option --gdb-port
usage: spike [host options] <target program> [target options]
Host Options:
  -p<n>                 Simulate <n> processors [default 1]
  -m<n>                 Provide <n> MiB of target memory [default 2048]
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
                          at base addresses a and b (with 4 KiB alignment)
  -d                    Interactive debug mode
  -g                    Track histogram of PCs
  -l                    Generate a log of execution
  -h                    Print this help message
  -H                 Start halted, allowing a debugger to connect
  --isa=<name>          RISC-V ISA string [default RV64IMAFDC]
  --pc=<address>        Override ELF entry point
  --ic=<S>:<W>:<B>      Instantiate a cache model with S sets,
  --dc=<S>:<W>:<B>        W ways, and B-byte blocks (with S and
  --l2=<S>:<W>:<B>        B both powers of 2).
  --extension=<name>    Specify RoCC Extension
  --extlib=<name>       Shared library to load
  --rbb-port=<port>     Listen on <port> for remote bitbang connection
  --dump-dts  Print device tree string and exit
```

I am unfamiliar with rbb. Is it some kind of debugging software like gdb? And why was support for gdb removed?

ibolton, 2017-06-05:
Hi, TheThirdOne. They took out the GDB Server a few months back on master branch.

timsifive, 2017-06-05:
Right.
In the old implementation gdb would connect directly to spike, speaking
gdb's protocol.
In the new implementation, spike exposes a fake JTAG device using
OpenOCD's remote bitbang (rbb) protocol. Now to debug spike you use gdb to
OpenOCD to spike.
This was changed to make spike reflect a real RISC-V target more
accurately. It also has the benefit that now we can test the OpenOCD code
without real hardware, which allows for automated testing etc.

I just got back from vacation, and right now my setup isn't working right.
Once I've got that figured out I will update the README. The OpenOCD config
file you want is
https://github.com/riscv/riscv-tests/blob/master/debug/targets/spike/openocd.cfg

Tim

Ma124, 2020-07-10:
The documented way to attach GDB still doesn't work for me.

<details>
<summary><code>riscv-openocd</code> doesn't recognize <code>remote_bitbang</code></summary>
<code><pre>
Open On-Chip Debugger 0.10.0+dev-01227-g8b929c649 (2020-07-10-19:24)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Error: The specified debug interface was not found (remote_bitbang)
The following debug adapters are available:
1: ftdi
2: usb_blaster
3: ft232r
4: presto
5: usbprog
6: openjtag
7: jlink
8: vsllink
9: rlink
10: ulink
11: arm-jtag-ew
12: hla
13: osbdm
14: opendous
15: aice
16: cmsis-dap
17: kitprog
18: xds110
19: st-link
</pre></code>
</details>

<details>
<summary>And plain OpenOCD does not know RiscV</summary>
<code><pre>
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
embedded:startup.tcl:21: Error: Unknown target type riscv, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11,ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx,  or quark_d20xx
in procedure 'script'
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "spike.cfg", line 9
in procedure 'ocd_bouncer'
at file "embedded:startup.tcl", line 21
</pre></code>
</details>

<details>
<summary>Given the documented config</summary>
<code><pre>
interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 9824

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10e31913

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

gdb_report_data_abort enable

init
halt
</pre></code>
</details>

TheThirdOne, 2020-07-10:
I am going to consider this solved for me. The readme has been updated to not suggest using gdb-port. Issues with OpenOCD not working correctly don't need to be kept here.

closed 2020-07-10 by TheThirdOne

timsifive, 2020-07-10:
To get remote bitbang support in OpenOCD, you need to run configure with `--enable-remote-bitbang` before building.
I'd accept a change https://github.com/riscv/riscv-openocd that makes that the default.