riscv-software-src/riscv-isa-sim #106: ../riscv/gdbserver.cc:352:9: error: statement will never be executed [-Werror=switch-unreachable]

promach, 2017-06-01:
I am having a problem building the toolchain. 

`[promach@localhost riscv-tools]$ ./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-fesvr/build directory
Configuring project riscv-fesvr
Building project riscv-fesvr
Installing project riscv-fesvr

Removing existing riscv-isa-sim/build directory
Configuring project riscv-isa-sim
Building project riscv-isa-sim
../spike_main/disasm.cc: In constructor ‘disassembler_t::disassembler_t(int)’:
../spike_main/disasm.cc:257:1: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 disassembler_t::disassembler_t(int xlen)
 ^~~~~~~~~~~~~~
../riscv/gdbserver.cc: In member function ‘virtual bool halt_op_t::perform_step(unsigned int)’:
../riscv/gdbserver.cc:352:9: error: statement will never be executed [-Werror=switch-unreachable]
         gs.tselect_valid = false;
         ^~
cc1plus: all warnings being treated as errors
make: *** [Makefile:319: gdbserver.o] Error 1
[promach@localhost riscv-tools]$ `

I have googled but no-one has the same error. I suppose -Werror is not encouraged in release code, [https://blog.flameeyes.eu/2009/02/future-proof-your-code-dont-use-werror/](url) or am I missing something ?

ibolton, 2017-06-05:
Hi promach. Which version of riscv-isa-sim do you have? On the master branch, the gdbserver.cc file has been removed from the repo (and the --gdb-port command-line option has gone as well).

promach, 2017-06-05:
@ibolton Thanks. It is solved now

closed 2017-06-05 by promach