riscv-software-src/riscv-isa-sim #122: Wrong PC for 32-bit mode

hauhsu, 2017-09-12:
I tried to build and run a simple 32-bit program with spike.
In debug mode, the printed PC seems sign-extended:

```
$ spike -d --isa=RV32G pk  a.out
:
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
:
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
:
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
:
core   0: 0x000000000000100c (0x0182a283) lw      t0, 24(t0)
:
core   0: 0x0000000000001010 (0x00028067) jr      t0
:
core   0: 0xffffffff80000000 (0x1f00006f) j       pc + 0x1f0
:
core   0: 0xffffffff800001f0 (0x00000093) li      ra, 0
...
```

Also, if I use `until` command to stop simulation when PC equals 0x80001234, I have to give command: `until pc 0 ffffffff80001234`. So it's not just a printing formate problem?

aswaterman, 2017-09-12:
Spike canonicalizes 32-bit addresses into 64 bits by sign-extending them.

hauhsu, 2017-09-12:
Thanks for fast reply!
But why use sign-extended address? Shouldn't an address be unsigned?

aswaterman, 2017-09-12:
The sign-extension approach makes two's complement comparisons (like SLT/BLT) behave as expected.

sam-falvo, 2017-09-12:
On Mon, Sep 11, 2017 at 7:26 PM, Andrew Waterman
<notifications@github.com> wrote:
> The sign-extension approach makes two's complement comparisons (like SLT/BLT) behave as expected.

To be fair, $80001234 in my mind is greater than $7FFF1234.

That said, to provide a more direct answer to the question, one reason
for having signed addresses is to allow semi-transparent migration
from 32-bit to 64-bit address spaces.  If you place I/O or other
resources at negative addresses, then widening the address space
merely involves reallocating address decoder bits (e.g., instead of
checking A31-A28, you'd use A63-A60).  This is a technique which the
Transputer series of processors used to great effect, so some prior
art on this exists.

-- 
Samuel A. Falvo II

aswaterman, 2017-09-12:
None of this is ISA-visible. It's just how the simulator is implemented.

hauhsu, 2017-09-12:
Thanks for the replies :)
I'll close this issue.

closed 2017-09-12 by hauhsu