riscv-software-src/riscv-isa-sim #124: How does spike deal with input? I want to script it.

vanjoe, 2017-09-13:
I've been trying to script spike control, but it doesn't seem to play nice with commands piped into it. Is there a way to get around that?

hauhsu, 2017-09-14:
Spike uses stderr as input in debug mode (not sure why).
I modified the code in `riscv/interactive.cc`  as issue #87 described.
Find the line `std::string s = readline(2);` and modify it to 
`std::string s = readline(0);`
Recompile spike.
Then I can write debug commands in a file: 
(cmd.txt)
```
until pc 0 0x0000000080007954
reg 0 t0
q
```

To execute those command:
`$ spike -d my.elf < cmd.txt`

vanjoe, 2017-09-14:
How is that even possible? Isn't stderr write only?

evancox10, 2017-09-15:
I just ran into this today. This is strange and should be fixed!

aswaterman, 2017-09-15:
After doing some archaeology, reading from stderr handles the case of the host and target simultaneously wishing to use stdin.  The target machine being able to use stdin is essential for running posix programs on ```pk```.

I guess you could make it so passing the ```-d``` flag implies you don't need target-machine stdin, so in that case, you can use stdin instead of stderr for the host console.  But that would break ```xspike``` (which spawns an xterm to have a separate console for host and target), so the solution's more complicated than that.

vanjoe, 2017-09-15:
Is there another way to script spike? What I really want is to get instruction traces, with ideally register values after each instruction.

ccelio, 2017-09-16:
Does the existing commit log capability fit your instruction trace requirements?

https://github.com/riscv/riscv-isa-sim/blob/master/config.h.in#L70
https://github.com/riscv/riscv-isa-sim/blob/master/riscv/execute.cc#L18-L33

evancox10, 2017-09-18:
I just found that after my last post, I'm going to recompile with it on and give that a try. Also, using STDERR for interactive control makes more sense, considering that STDIN/STDOUT are reserved for the emulated program.

Something like GDB's `-x` argument or `source` command would also be useful, however.

evancox10, 2017-09-26:
I've got the commit log working and was able to modify its output for what I needed. However, it would be nice to be able to turn the output on or off. Any reason why commit log and histogram are compile time options, and not run-time switches? Maybe performance issues in the critical fetch/decode/execute loop for the instruction .h files that get included?

I tried to install two versions of the binary by keeping two copies of $RISCV/bin/spike, but this didn't work. I get the same behavior when running spike-nolog and spike-log. Is it calling some other code/executable in another location?

asynts, 2022-12-07:
For people who like me spend several hours trying to get this working to then realize that there is an `--debug-cmd` flag. Just use the `--debug-cmd` flag.

<details>
<summary>My complete setup</summary>

```none
# run.sh

[[ -d build/ ]] && rm -r build/ ||:
mkdir build/

riscv64-elf-as -c simple.S -o build/simple.o
riscv64-elf-ld -T simple.x -o build/simple build/simple.o
spike --isa=rv32i -m0x80000000:0x10000 -d --debug-cmd=commands.txt build/simple
```

```assembly
```

</details>

manasabatchu, 2024-03-14:
I did riscv toolchain modification for crypto instructions and I am able to generate executable files and I changed spike according to the requirements for crypto instructions but when I am running executable files in the riscv spike I am getting tohost error how can I fix it