riscv-software-src/riscv-isa-sim #125: read csr register with spike

esolera, 2017-09-20:
Im not sure if I can do this with spike. I make a hello world and compile in  32 bits like you see below and I tried to measure the machine cycles:

#include <stdio.h>

#define read_csr(reg) ({ unsigned long __tmp; \
  asm volatile ("csrr %0, " #reg : "=r"(__tmp)); \
  __tmp; })
#define CSR_CYCLE 0xc00
#define CSR_TIME 0xc01
#define CSR_INSTRET 0xc02
#define CSR_MCYCLE 0xb00

int main(void) { 
long cycles;
printf("Hello world!\n");
cycles=read_csr(mcycle);
 printf(" %ld\n", cycles);
 return 0; 

}

But when a run it he give the next message:
  Hello world!
z  00000000 ra 000101d8 sp 7fbe5d80 gp 000151a8
tp 00000000 t0 000103c8 t1 00012cd8 t2 00000000
s0 7fbe5da0 s1 00000000 a0 0000000a a1 00014a28
a2 0000000d a3 00000000 a4 00000000 a5 0000000a
a6 0000000f a7 00000040 s2 00000000 s3 00000000
s4 00000000 s5 00000000 s6 00000000 s7 00000000
s8 00000000 s9 00000000 sA 00000000 sB 00000000
t3 00000000 t4 00000000 t5 00000000 t6 00000000
pc 000101d8 va b00027f3 insn       b00027f3 sr 80046020
An illegal instruction was executed!

My questions is : Is posible test and see the csr in spike?

aswaterman, 2017-09-20:
```mcycle``` isn't available in user mode, but ```cycle``` is.

qian-gu, 2022-08-23:
So is there a way to read `mcycle` in spike? I have tried following code to switch into machine mode, but got an error saying:

> bad syscall #0!

```c
asm __volatile__("ecall\n" : : );
cycles = read_csr(mcycle);
asm __volatile__("mret\n" : : );
```

aswaterman, 2022-08-23:
Try reading the cycle CSR instead of mcycle (and remove the ECALL and MRET).

qian-gu, 2022-08-24:
Actually I have a RV32 core which only implement M mode, following code has been test passed on a FPGA board:

```c
int rdcycle() {
    int tmp = 0;
    asm __volatile__ ("csrr %[dst01], mcycle\n" : [dst01]"=r"(tmp) : : );
    return tmp;
}
```

and I implement another version `rdcycle()` function for spike :

```c
int rdcycle() {
    int tmp = 0;
    asm __volatile__ ("rdcycle $[dst01]\n" : [dst01]"=r"(tmp) : : );
    return tmp;
}
```

I wonder whether it's possible to read `mcycle` in spike so that keeping codes on FPGA board and spike uniform.

akashtenkale03, 2023-01-09:
i've tried it cycle, mcycle and for time also (all are csr). but every time i got a single specific value, just like they are hardcoded. for time : 0xc01027f3, cycle : 0xc00027f3. Does anyone else got the same values or different and how to overcome it ?

jerryz123, 2023-01-09:
It looks like you are reading the instruction bits of the instructions themselves, rather than the value written by the instructions. 

There are plenty of examples of code which read these counters and print the values. Try looking at the benchmarks in the riscv-tests repository.

akashtenkale03, 2023-01-10:
Yesss, i forget to apply the --log-commits flag.

ekliptik, 2023-10-25:
I believe this issue could be closed. If it is truly necessary to have cycle CSR available, the users can be directed to using the RISC-V proxy kernel which has a trap handler, and translates cycle CSR reads to mcycle CSR reads
https://github.com/riscv-software-src/riscv-pk

davidcastells, 2024-06-25:
I would like to insist in this question from another perspective. In spike interactive debug (-d) you can examine registers of the processor (with reg), but can you examine CSRs with any interactive command ??

aswaterman, 2024-06-25:
Yes, just use the name of the CSR, as in `reg 0 mstatus`.

hirooih, 2024-08-16:
I found this issue ticket because I had similar problem that the latest spike caused an illegal instruction exception on `rdcycle`.
The above thread did not solve the problem directly, but I found the cause.

Since 53a3002 we have to add something like `--isa=rv32imc_zicntr` option on spike.

I leave this information here for those who will have a similar problem.