riscv-software-src/riscv-isa-sim #131: invalid store to debug module: 8 bytes at 0x0000000000004000

karimhasebou, 2017-10-06:
I have compiled my application and linked it with an RTOS. Everytime I try to execute it on spike(am not using pk)  I keep getting an error " terminate called after throwing an instance of 'trap_store_access_fault' " . Am also using isa=rv32i if that matters

luismarques, 2017-10-06:
You have to use the proxy kernel, since it's the pk that maps the address space properly. Spike only configures the address space for the pk to run, so unless your bare metal app happens to match the memory ranges of the pk it won't run. I know that's not the case for you because my program also has an entry point at 0x2000 and it needs the pk to run in spike. Also, be sure that you use a 32-bit pk. In the end I just coded my own RISC-V simulator, since spike's reliance on pk makes it inconvenient for several use cases.

karimhasebou, 2017-10-06:
@luismarques thank you for helping. By any chance did you get an error -m32 flag doesn't exist when you tried to compile pk as 32 bit? am using  a prebuilt toolchain provided by SiFive.

luismarques, 2017-10-06:
@karimhasebou I had a lot of trouble compiling a 32-bit pk on macOS. Eventually I just gave up, compiled a 32-bit pk on Ubuntu and copied it back to the mac. Since pk is an RISC-V ELF executable that's fine. But spike and related repos have changed a bit since then, so YMMV.

karimhasebou, 2017-10-08:
@luismarques  Is there anyway to find out the address space spike uses for pk so I can compile my code for that configuration. From what I read, debugging with pk is challenging and therefore I'd like to be able to use GDB directly along with openocd

evancox10, 2017-10-13:
I'm also trying to run spike in baremetal, but running into issues with store access faults. Apparently there is a debug/boot ROM placed at 0x200, spike's starting PC. If you try to load an ELF with the entry point set here, then spike calls sim_t's write_chunk function to write to this area. But since this region is read only, you get a trap_store_access_fault.

I don't know how far this region extends. I'm working my way through the spike code to figure out what it's actually doing.

Also, there is an open bug in the configure scripts for building a 32-bit pk. See [this issue](https://github.com/riscv/riscv-pk/issues/56) for the fix. That worked for me on RHEL6. No idea about OSX.

This is the best documentation I've found on spike's internal operation:
https://github.com/poweihuang17/Documentation_Spike

1934224526, 2018-03-12:
can spike run without pk in Bare metal mode? There is a error the same as above when I run it without pk, how to solve the problem? Thank you.

timsifive, 2018-03-12:
Yes. The debug test suite does this. An example command line is:
`+ /opt/riscv/bin/riscv64-unknown-elf-gcc -g -march=rv32imac -mabi=ilp32 programs/checksum.c programs/tiny-malloc.c programs/infinite_loop.S -DDEFINE_MALLOC -DDEFINE_FREE programs/entry.S programs/init.c -DNHARTS=1 -I ../env -march=rv32imafdc -T targets/RISC-V/spike32.lds -nostartfiles -mcmodel=medany -DXLEN=32 -o spike32_checksum-32`

The referenced source and link files are in https://github.com/riscv/riscv-tests/tree/master/debug

1934224526, 2018-03-13:
@timsifive ,I have tried your method, but it still doesn't work.The error message as follow:
terminate called after throwing an instance of 'trap_load_access_fault'
I did't add the pk directive when I run spike

timsifive, 2018-03-13:
@1934224526, then I'm not sure what's going on. All I can say is that bare metal programs are definitely possible. I use them all the time.