riscv-software-src/riscv-isa-sim #132: tohost and fromhost symbols not in ELF

karimhasebou, 2017-10-07:


luismarques, 2017-10-07:
`tohost` and `fromhost` are used to send requests to the simulator host. They are present in the proxy kernel, which uses it to forward requests for system calls like `write` to your actual OS. You should provide a description in your issue of what your problem is. AFAIR this is a warning and not an error so you can ignore it. If that's inconvenient, then just define those symbols. Or, better yet, implement to some extent `tohost` and `fromhost` to allow your simulated programs to actually communicate with your host OS (e.g. print to stdout).

karimhasebou, 2017-10-07:
@luismarques I thought that warning was the reason gdb is not working however I was able to get a simple main function to compile and gdb was able to debug.

IC-Dream, 2018-11-08:
Hi, @luismarques,

I also enounter this issue, and the spike hangs and can't exit. so do you know the flow of spike exit, for the ELF form riscv/riscv_tests, it would enter a loop of write_tohost, the spike exit after loop many times.

Thanks
Dream

luismarques, 2018-11-10:
@IC-Dream I think you are going to have to rephrase your issue a bit more clearly.

IC-Dream, 2018-11-12:
@luismarques thanks a lot， i added the tohost and fromhost symbols in .ld files and jump to write "1" to tohost in the end of test case.  the spike can exit successfully.

JerryShih, 2019-08-23:
I have the same issue, but it's due to the wrong libmetal library. When I link the correct libmetal library, I could use spike without any problem.

The libmetal library with spike supports.
https://github.com/sifive/freedom-metal/tree/htif

isubasinghe, 2022-07-08:
@luismarques do you know if spike can print terminal output this way?

luismarques, 2022-07-08:
You can see an example of a completely self-contained bare-metal program that outputs to the console using the HTIF interface (`tohost`, `fromhost`) here: <https://github.com/lowRISC/epic-c-example>. So, yes, @isubasinghe it can print terminal output. See the `htif.c` file, as well as the `util.c` file (contains a `print` function).

isubasinghe, 2022-07-08:
Thank you so much for your quick response @luismarques , this is super useful for me.

svasekar, 2024-07-19:
> @luismarques thanks a lot， i added the tohost and fromhost symbols in .ld files and jump to write "1" to tohost in the end of test case. the spike can exit successfully.
@IC-Dream  will you please mentioned in detail how did you solved the issues,as I am facing same issue 
my linker has to host and fromhost,  do I need to add anything in program i.e. in .S file?
linker.ld
OUTPUT_ARCH( "riscv" )
ENTRY(rvtest_entry_point)

SECTIONS
{
  . = 0x80000000;
  .text.init : { *(.text.init) }
  . = ALIGN(0x1000);
  .tohost : { *(.tohost) }
  . = ALIGN(0x1000);
  .text : { *(.text) }
  . = ALIGN(0x1000);
  .data : { *(.data) }
  .data.string : { *(.data.string)}
  .bss : { *(.bss) }
  . = ALIGN(8);
  PROVIDE(tohost = .);
  . +=8;
  PROVIDE(fromhost = .);
  . +=8;
  _end = .;
}

however, while loading elf, in SPIKE, i get the same warning and SPIKE hangs.