riscv-software-src/riscv-isa-sim #136: how to simulate new instruction in spike

thecakim, 2017-10-14:
Hi ... I created a <instruction name> .h which describes the behavior of the desired instruction in ../riscv/insns.
In ../riscv-tools/riscv-opcdes, I added the opcode like other forms. I also found that opcode does not overlap with $ make install.

But how do I make the new instruction as input to the spike?
If I modify the assembly code, it's up to gcc to change it to an executable, but gcc does not know what I have added to the instruction.

Q: Is there a way to execute instruction I have added through spike, other than modifying gcc?

sherrbc1, 2017-11-14:
You would need to modify the assembler unless you wanted the GCC compiler to emit assembly using your new instruction directly (though I suspect this would be difficult). I would write either inline assembly or asm directly, then modify the assembler to output the appropriate machine code when parsing the new instruction.

seldridge, 2017-11-14:
The super lazy way is to write inline assembly that emits raw data which will be decoded as your new instructions. It's kludgy, but you can get pretty far by then wrapping this up with macros, e.g., [the non-standard approach for emitting RoCC instruction](https://github.com/IBM/rocc-software/blob/master/src/xcustom.h#L53). 

The clean way to do this is with "instruction pseudos", but I don't think anyone has had bandwidth to fully implement that. See: https://github.com/riscv/riscv-binutils-gdb/issues/38.

Long original discussion here: https://github.com/riscv/riscv-gnu-toolchain/issues/190