riscv-software-src/riscv-isa-sim #139: can't write MEIE of MIE csr

hustcpp, 2017-10-19:
Hi,

Seems MIP_MEIP bit of MIE is not writable, is this correct?

https://github.com/riscv/riscv-isa-sim/blob/a91d9f7d89abd6cda6fafb7b5e7cacf4b3590c29/riscv/processor.cc#L309

https://github.com/riscv/riscv-isa-sim/blob/a91d9f7d89abd6cda6fafb7b5e7cacf4b3590c29/riscv/processor.cc#L357

Thanks.

aswaterman, 2017-10-19:
From the spec:  "The MEIP field in mip is a read-only bit that indicates a machine-mode external interrupt is pending. MEIP is set and cleared by a platform-specific interrupt controller, such as the standard platform-level interrupt controller specified in Chapter 7."

hustcpp, 2017-10-19:
Hi aswaterman, yes, the bit in mip is not writable. how about mie? Thanks.

aswaterman, 2017-10-19:
Ah, I misunderstood your original comment because MIE doesn't have an MEIP bit (it's called MEIE; the code is misleading).

The MEIE bit is not writable in Spike because Spike does not model the platform-level interrupt controller.  Since MEIP is always 0, there is no reason to allow MEIE to be writable.

hustcpp, 2017-10-19:
Thanks, but I still consider it not align with Spec. at least the read/write behavior of MIE is different.

And 3rd party could implement a PLIC and connect it MIP. then they will find external IRQ can't be enabled

aswaterman, 2017-10-19:
I respectfully disagree.

If a user of Spike implements a PLIC, they should make MEIE writable at that time.

closed 2017-10-19 by hustcpp