riscv-software-src/riscv-isa-sim #141: Commit-Log builds broken due to Q extension

evancox10, 2017-10-19:
The WRITE_FREG macro in decode.h is broken for builds with commit log enabled (`-DRISCV_ENABLE_COMMITLOG`)

```
#ifndef RISCV_ENABLE_COMMITLOG
# define WRITE_REG(reg, value) STATE.XPR.write(reg, value)
# define WRITE_FREG(reg, value) DO_WRITE_FREG(reg, freg(value))
#else
# define WRITE_REG(reg, value) ({ \
    reg_t wdata = (value); /* value may have side effects */ \
    STATE.log_reg_write = (commit_log_reg_t){(reg) << 1, wdata}; \
    STATE.XPR.write(reg, wdata); \
  })
# define WRITE_FREG(reg, value) ({ \
    freg_t wdata = freg(value); /* value may have side effects */ \
    STATE.log_reg_write = (commit_log_reg_t){((reg) << 1) | 1, wdata.v}; \
    DO_WRITE_FREG(reg, wdata); \
  })
#endif
```

The type of wdata is now `float128_t`, which is typdef'd in softfloat_types.h to struct { uint64_t v[2]; }`. So, this code is trying to convert from uint64_t* to uint64_t, which isn't allowed.

To get things compiling you can change `wdata.v` to `wdata.v[0]`. However, I'm not sure if this actually gets you the right data for f32/64 types. I'm not doing any floating point work so can't test this easily. A simple solution for these types is probably available using the box/unbox convenience macros in decode.h .

However, to actually log float128_t writes in the commit log, more changes would be needed to expand commit log to support the bigger type.

aswaterman, 2017-10-19:
@ccelio I guess only printing the lower 64 bits is OK, since nothing that uses the commit log (besides spike) supports the Q extension?  Or should we actually print all 128 bits?

ccelio, 2017-10-19:
Thanks for the report @evancox10. 

although annoying, we should do things right and go with the least surprising behavior. I assume we can't know (max XLEN FLEN) and print based on that? Then we should do 128b. If space is an issue one could pipe to a compressor.

evancox10, 2017-10-19:
@aswaterman The commit log is useful as a general purpose tool for anyone implementing their own CPU or functional units. IMO we can't assume that just because the publicly available/known tools do not use it, that therefore nothing does. I'm sure that someone designing a 128-bit FPU from scratch would appreciate known-good output from the reference model (ie spike).

BTW, when operating in 32-bit mode, spike already sign-extends registers and the PC to 64 bits for display output, whether that's commit log, interactive display, etc (see #66 ). This is kind of annoying, and I would hope this gets fixed along with any changes to switch between 64/128 floating point. I would wager there are far more people interested in rv32 than quad FP.

Also, there's a build server somewhere right? You may want to have the server do commit-log builds of spike as part of its test. Maybe the other common compile-time options too.

aswaterman, 2017-10-19:
OK.

I'm going to leave it to @ccelio to deal with the RV32 issue.

Agreed re: regression tests; we just haven't had the time.

aswaterman, 2017-10-20:
Both issues resolved by #143

closed 2017-10-20 by aswaterman