riscv-software-src/riscv-isa-sim #147: --isa=rv32imaf fails to enable floating point support

asb, 2017-10-30:
It looks like there are multiple issues here, but the most basic one seems to be that floating point instructions like fcvt.s.w and fmv.x.w are trapping even with --isa=rv32imaf.

I found this while investigating a rather odd test failure. Consider the program:
```
$ cat t.c
#include <stdio.h>
#include <stdlib.h>

float s2f(int s) {
  return s;
}

int main() {
  if (s2f(0x80000000) != -2147483648.0)
    abort();
  return 0;
}
```
This (incorrectly) aborts on spike when running a binary compiled with -march=rv32imaf -mabi=ilp32.

Looking at the output of `spike -l`, I can see that the floating point instructions triggered traps. The fact that the emulation routines failed to produce the correct result is most likely another issue, but lets start here.

Tool versions:
* riscv-fesvr matching the version pinned in the riscv-tools repo (f683e01)
* riscv-isa-sim matching the version pinned in the riscv-tools repo (3a4e893), configured with `--with-isa=rv32imaf`
* riscv-pk matching the version pinned in the riscv-tools repo (2dcae92)

To reproduce:
```
riscv32-unknown-elf-gcc -O -march=rv32imaf -mabi=ilp32 t.c
spike --isa=rv32imaf pk a.out; echo $?
spike -l --isa=rv32imaf pk a.out &> log
# View log to observe traps for fcvt, fmv
```

aswaterman, 2017-10-30:
It looks like this a limitation of pk; it only supports FD or no floating-point at all: https://github.com/riscv/riscv-pk/blob/master/machine/fp_asm.S#L5

Since that error is not being triggered, I guess you're compiling pk without the f extension.  So it's disabling the FPU on boot: https://github.com/riscv/riscv-pk/blob/master/machine/minit.c#L53

I think the main difficulty of supporting this use case is that the floating-point emulation code only works when the floating-point registers are either wide enough to hold all operands, or when they don't exist at all.  In other words, the case of emulating double-precision operations with a single-precision regfile (which would require splitting the data between registers and memory) is not supported.  I guess we could support the F-only case but reject the option to emulate the D extension in this case.

asb, 2017-10-31:
Thanks for the insight Andrew. You're right that I'm compiling pk with a -march=rv32ima defaulting GCC. I hadn't imagined pk would decide to disable the FPU in that case. It's not immediately clear why that is desirable, but I'm obviously not very familiar with pk internals.

Closing this issue and following up with a riscv-pk bug report on the incorrect result.

closed 2017-10-31 by asb

panda1628, 2021-01-12:
I run into similar issue.

spike -l --isa=rv32imaf --log-commits fadd.s.bfd

core   0: 0xffffffffc0500310 (0x0021f1d3) fadd.s  ft3, ft3, ft2
core   0: exception trap_illegal_instruction, epc 0xffffffffc0500310
core   0:           tval 0x000000000021f1d3
core   0: >>>>  _trap_handler

Do I miss anything? Thx