riscv-software-src/riscv-isa-sim #150: `spike-dasm` is disassembling `li` as `jr`

mwachs5, 2017-11-02:
riscv64-unknown-elf-objdump output is this:
```
20000000 <_prog_start>:
20000000:       4081                    li      ra,0
20000002:       4101                    li      sp,0
20000004:       4181                    li      gp,0
20000006:       4201                    li      tp,0
20000008:       4281                    li      t0,0
2000000a:       4301                    li      t1,0
```

Trace file matches the above:
```

aswaterman, 2017-11-02:
Looks like Palmer got to this a few months ago in 7f746b7c2f27f6271640efaeb2cde86fd1b86957

closed 2017-11-02 by aswaterman

mwachs5, 2017-11-02:
hmm weird, i thought I just bumped riscv-isa-sim up to something pretty recent... hopefully I didn't just screw that up.