riscv-software-src/riscv-isa-sim #164: Access to MSIP within CLINT causes system exceptions from Machine-mode

sherrbc1, 2017-12-01:
A Machine-mode access to the MSIP register for hartid=0 within the CLINT periperhal will trigger a system exception with `mcause`=7 (Store/AMO access fault). 

`*((uint32_t *) CLINT_BASE + CLINT_MSIP_OFFSET) = 1;`

And a Machine-mode access to the MSIP register for hartid=1 will trigger a system exception with `mcause`=6 (Store/AMO address misaligned). 

`*((uint32_t *) CLINT_BASE + CLINT_MSIP_OFFSET) = 1;`

Where 

```
#define CLINT_BASE                (0x2000000)
#define CLINT_H0_MSIP_OFFSET      (0)
#define CLINT_H1_MSIP_OFFSET      (4)
```

Relevant clint definition: https://github.com/riscv/riscv-isa-sim/blob/master/riscv/clint.cc#L9-L10

sherrbc1, 2017-12-01:
The first problem turned out to be an access length issue in the CLINT implementation. The `msip` register is only 4 bytes wide, so accesses of any other length will fail with an access fault exception. 

https://github.com/riscv/riscv-isa-sim/blob/master/riscv/clint.cc#L42

aswaterman, 2017-12-01:
I don't think it's a Spike bug.  You should examine the assembly code you're generating and make sure it's doing what you expect.  I see two problems with the C code:

- ```(uint32_t*)base + 4``` isn't what you want: you should either do ```(uint32_t*)(base + 4)``` or ```(uint32_t*)base + 1```.
- You should mark the pointer volatile.

sherrbc1, 2017-12-01:
The pointer arithmetic is a mistake indeed, but the original problem was related to the CLINT implementation expecting MSIP accesses of width. I wrote the code assuming 64 bit accesses while the example I provided used 32. Aside from the arithmetic error you mention, the code works as expected (for 32 bit accesses).

aswaterman, 2017-12-01:
Ah, I see.

The CLINT spec says only 32-bit accesses are supported.

sherrbc1, 2017-12-01:
Only `msip` is defined as 32-bit. Both `mtime` and `mtimecmp` are 64-bit.
https://github.com/riscv/riscv-isa-sim/blob/master/riscv/devices.h#L69-L71

aswaterman, 2017-12-01:
Sorry... I meant that MSIP bits only support 32-bit accesses.

closed 2017-12-01 by sherrbc1