riscv-software-src/riscv-isa-sim #169: Does spike have an implicit watchdog timer ?

LeeMooreImperas, 2018-01-23:
Hi All
I have been running spike on one of the riscv-tests (dated for the 1.10Priv/2.2User spec)
the test is rv32mi-p-illegal
When run on spike it enters an infinite loop
```
core   0: 0xffffffff8000015c (0x0000006f) j       pc + 0x0
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
```

However the loop eventually breaks out due to an interrupt
```
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
core   0: exception interrupt #1, epc 0xffffffff8000015c
core   0: Executed 1376 times
core   0: 0x0000000080000304 (0xe5dff06f) j       pc - 0x1a4
3 0x0000000080000304 (0xe5dff06f)
core   0: 0xffffffff80000160 (0x30541073) csrw    mtvec, s0
3 0xffffffff80000160 (0x30541073)
core   0: 0xffffffff80000164 (0x30315073) csrwi   mideleg, 2
3 0xffffffff80000164 (0x30315073)
```

Where is the behavior of this interrupt generation defined ?
I have looked through the source of the simulator, but nothing obvious appears

Thx
Lee

neelgala, 2018-01-23:
If the cause is a timer interrupt then you should probably look at the file
clint.cc <https://github.com/riscv/riscv-isa-sim/blob/master/riscv/clint.cc> .
Line 70 is where the interrupt bit MTIP is being set.

LeeMooreImperas, 2018-01-23:
@neelgala 
I think my issue is that I do not know if the cause is a timer interrupt
The symptom is that an executable is running which is stuck in an infinite loop (jump to self)
the loop is exited by "exception interrupt #1"

I have no idea what "exception interrupt #1" is, how or where it is defined.
is this implicit in the simulator, or under user control ?
In either case, what are the conditions for this exception to occur ?

Thx
Lee

aswaterman, 2018-01-23:
Interrupt 1 is a supervisor software interrupt (see Table 3.6 in priv spec).  If you look at rv32mi/illegal.S, you'll see that the test itself is setting up that interrupt.

LeeMooreImperas, 2018-01-24:
@aswaterman 
Hi Andrew, maybe I did not phrase my question very well.
What I cannot work out is what (in the simulator) causes the exception to occur ?

I understand that illegal.S has setup the registers to receive and handle an interrupt/exception.
My question is - how did this exception/interrupt get generated ?
usually in a platform an external interrupt is generated by some event for example a timer expiring.

so what event caused the following to be initiated ?
`core   0: exception interrupt #1, epc 0xffffffff8000015c`

is there an (undocumented) builtin timer in the simulator which expires causing this interrupt ?
I cannot get any visibility for the cause of the interrupt generation, 

Thx
Lee

LeeMooreImperas, 2018-01-24:
@aswaterman 
Hi Andrew,
Thanks, I now see what is happening.
We have built the simulator to enable register tracing, and when we run this example we see the following 
```
core   0: 0xffffffff80000158 (0x30046073) csrsi   mstatus, 8
3 0xffffffff80000158 (0x30046073)
core   0: 0xffffffff8000015c (0x0000006f) j       pc + 0x0
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
3 0xffffffff8000015c (0x0000006f)
... < repeat a total of 4982 times>
core   0: exception interrupt #1, epc 0xffffffff8000015c
core   0: Executed 1376 times
core   0: 0x0000000080000304 (0xe5dff06f) j       pc - 0x1a4
3 0x0000000080000304 (0xe5dff06f)
core   0: 0xffffffff80000160 (0x30541073) csrw    mtvec, s0
3 0xffffffff80000160 (0x30541073)
```

I think what confused me is that we see a long pause (approx 5000 lines) between the enable interrupt, and interrupt taken. 
These are nothing to do with additional instructions being executed.

Can I ask why we see many repeted lines of 
`3 0xffffffff8000015c (0x0000006f)`
between the interrupt enable, and raising of the exception ?

thx
Lee

aswaterman, 2018-01-24:
For reasons of simulator performance, Spike doesn't check for interrupts on every instruction.  ```sim_t::INTERLEAVE```, which is set to 5000, controls how often it checks for interrupts or switches between threads.