riscv-software-src/riscv-isa-sim #170: Misaligned loads and stores

LeeMooreImperas, 2018-01-23:
In the 2.2 User specification, says (page 19)
```
The base ISA supports misaligned accesses, but these might run extremely slowly
depending on the implementation. Furthermore, naturally aligned loads and stores are guaranteed
to execute atomically, whereas misaligned loads and stores might not, and hence require additional
synchronization to ensure atomicity.
```

However running one of the riscv-tests to exercise misalignd loads/stores I get an exception trap_load_address_misaligned
```
core   0: 0xffffffff800000cc (0xf3840413) addi    s0, s0, -200
3 0xffffffff800000cc (0xf3840413) x 8 0xffffffff80002000
...
core   0: 0xffffffff800000d8 (0x00141003) lh      zero, 1(s0)
core   0: exception trap_load_address_misaligned, epc 0xffffffff800000d8
core   0:           badaddr 0xffffffff80002001
```

Why am I seeing an exception for a misaligned load ?

Thx
Lee

sorear, 2018-01-23:
Misaligned stores are actually a Linux ABI requirement which is in the user-level spec for historical reasons.  Implementations can implement them in hardware or via a trap to M-mode; spike (and rocket-chip) chooses the latter.

LeeMooreImperas, 2018-01-23:
@sorear 
Thanks for the update.

I'm not sure of the relevance of the statement
`Misaligned stores are actually a Linux ABI requirement which is in the user-level spec for historical reasons. `
When referring to the ISA spec, but your second statement agrees with what we are seeing in the simulator, thanks for the clarification.
In fact most physical implementations we have seen, appear to match with a trap to M-Mode for misaligned load/store

Thx
Lee

AntonKrug, 2018-02-23:
Hi @LeeMooreImperas 

If you are building it from C then would you give a try to the "-mtune=size"? I think it has one of side effect that should tell the compiler not to use the "slow misaligned access" and not depend on trap handler to handle them.

Depending on the compiler version this compiler argument could probably help as well "-mstrict-align"

AntonKrug, 2019-04-24:
Hi @LeeMooreImperas,

Because you didn't respond, it is safe to assume that this is resolved and the issue can be closed?

eroom1966, 2019-04-24:
Hi Anton
Sorry for not responding, yes please go ahead and close
Thx
Lee

AntonKrug, 2019-04-24:
Hi @eroom1966,

I'm not owner/admin of this repository so I can't do it (I just noticed this issue is still opened when I was looking for something else). Your other account @LeeMooreImperas should be able to close it as you are the one who opened it.

closed 2019-04-24 by aswaterman

RogerYaozhenyu, 2020-12-01:
@AntonKrug   hi,  i am encountering the same issue. 。
(gdb) info registers mtval
mtval          0x3000d55d	805360989
(gdb) info registers mcause
mcause         0x6	6
(gdb) 

typedef struct 
{
    uint8_t inUse:1;
    uint8_t idx :7;
    taskPoolJob_Handle_t  job;
} __attribute__((packed)) taskJob_Rep ;
if i modified to belowd definition (aligned) this issue is gone.   
typedef struct 
{
    uint8_t inUse:1;
    uint8_t idx :7;
    taskPoolJob_Handle_t  job;
} taskJob_Rep ;

and i checked the compile flags .there is no mtune or strict-align   flags.  

`zhenyuy@chendunw-ThinkCentre-E76a-06:~/vulcan/node_nmci_e31_pcpk9_01$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/zhenyuy/vulcan/riscv64-unknown-elf-gcc-8.3.0-2020.04.1-x86_64-linux-ubuntu14/bin/../libexec/gcc/riscv64-unknown-elf/8.3.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/build/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --host=x86_64-linux-gnu --prefix=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-8.3.0-2020.04.1-x86_64-linux-ubuntu14 --with-pkgversion='SiFive GCC 8.3.0-2020.04.1' --with-bugurl=https://github.com/sifive/freedom-tools/issues --disable-shared --disable-threads --enable-languages=c,c++ --enable-tls --with-newlib --with-sysroot=/scratch/jenkins/workspace/tpp-freedom-tools/tpp03--build-binary-packages--parameterized/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-8.3.0-2020.04.1-x86_64-linux-ubuntu14/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=../riscv-gcc --with-system-zlib --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc CFLAGS=-O2 CXXFLAGS=-O2 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medany'
Thread model: single
`

what should i need to check now? please kindly give me 
some adivse. thanks.