riscv-software-src/riscv-isa-sim #190: Compressed shift instructions with zero shamt are hints on RV32/RV64 and should not trap

cliffordwolf, 2018-04-04:
I'm re-running the formal checks that compare riscv-formal with riscv-isa-sim and to my surprise there are new bugs in riscv-isa-sim that haven't been there last time I ran my checks..

Commit be0555d585b332fd0496affe559c0a5a4e7e5644 added checks that result in compressed shift instructions with zero shift amount to trigger an illegal instruction exception. That is a bug!

Compressed shift instructions with shamt=0 are hint instructions on RV32/RV64 and should behave as NOPs, not trigger an illegal instruction exception. I therefore recommend to revert commit be0555d585b332fd0496affe559c0a5a4e7e5644.

Excerpts from table 12.5 and table 12.6 of v2.2 of the ISA spec:

[image]

[image]

(cc @pmundkur)

aswaterman, 2018-04-04:
It looks like there's an ambiguity in the spec: the English says "For RV32C and RV64C, the shift
amount must be non-zero," which I suppose could be interpreted to mean "software that is not intending to use a HINT should not use a zero shift amount" but which the author of the patch interpreted as "these should trap" (see #181).

I agree with your interpretation and your proposed solution.

aswaterman, 2018-04-04:
Fixed, thanks Cliff.

closed 2018-04-04 by aswaterman