riscv-software-src/riscv-isa-sim #193: Multiple application on Spike Cores/harts

rajeshkmeena, 2018-04-06:
It is possible to run multicore/mulithart on spike using -p(num) and control the each hart/core by gdb .....

But, Is it possible to run multiple application (executable file) on multiple spike core/harts. If yes, then how .....??

I tried in following ways.....but failed....

  spike -l -p2 --rbb-port=9824 -m0x20000000:0x40000 test-64 rot13-64 
 
or
 
 spike -l -p2 --rbb-port=9824 -m0x20000000:0x40000 hartids=0 test-64  hartsids=1 rot13-64

Help please .....

moy, 2019-08-21:
Just to save others from the trouble of finding the file, `crt.S` is here: https://github.com/riscv/riscv-tests/blob/master/benchmarks/common/crt.S

The [multi-thread vector add](https://github.com/riscv/riscv-tests/tree/master/benchmarks/mt-vvadd) seems to be a good starting point, but although the `mt-` prefix, it actually seems to run only on one core because it links with the `crt.S` file you're quoting.

So I guess hacking `crt.S` to actually run `_init` for all cores would work. If anyone knows a place where one can find ready-to-use example, I'm interested. Otherwise, I guess I'll have to write it ;-).

powderluv, 2020-11-16:
did you write one ? I am looking for a simple example too.