riscv-software-src/riscv-isa-sim #194: `medeleg` doesn't support most exceptions

msuozzo, 2018-04-20:
Currently, QEMU supports a large number of delegable exceptions [1] ([src](https://github.com/riscv/riscv-qemu/blob/riscv-all/target/riscv/csr.c#L218-L233)) while spike only supports a handful [2] ([src](https://github.com/riscv/riscv-isa-sim/blob/master/riscv/processor.cc#L393-L399)). One notable exclusion is `ILLEGAL_INST`. Any reason for this disparity?

[1]: From riscv-qemu/targets/riscv/csr.c
```
static const target_ulong delegable_excps =
    (1ULL << (RISCV_EXCP_INST_ADDR_MIS)) |
    (1ULL << (RISCV_EXCP_INST_ACCESS_FAULT)) |
    (1ULL << (RISCV_EXCP_ILLEGAL_INST)) |
    (1ULL << (RISCV_EXCP_BREAKPOINT)) |
    (1ULL << (RISCV_EXCP_LOAD_ADDR_MIS)) |
    (1ULL << (RISCV_EXCP_LOAD_ACCESS_FAULT)) |
    (1ULL << (RISCV_EXCP_STORE_AMO_ADDR_MIS)) |
    (1ULL << (RISCV_EXCP_STORE_AMO_ACCESS_FAULT)) |
    (1ULL << (RISCV_EXCP_U_ECALL)) |
    (1ULL << (RISCV_EXCP_S_ECALL)) |
    (1ULL << (RISCV_EXCP_H_ECALL)) |
    (1ULL << (RISCV_EXCP_M_ECALL)) |
    (1ULL << (RISCV_EXCP_INST_PAGE_FAULT)) |
    (1ULL << (RISCV_EXCP_LOAD_PAGE_FAULT)) |
    (1ULL << (RISCV_EXCP_STORE_PAGE_FAULT));
```

[2]: From riscv-isa-sim
```
reg_t mask =
    (1 << CAUSE_MISALIGNED_FETCH) |
    (1 << CAUSE_BREAKPOINT) |
    (1 << CAUSE_USER_ECALL) |
    (1 << CAUSE_FETCH_PAGE_FAULT) |
    (1 << CAUSE_LOAD_PAGE_FAULT) |
    (1 << CAUSE_STORE_PAGE_FAULT);
```

aswaterman, 2018-04-20:
Illegal instructions generally shouldn’t be delegated because M-mode needs to intercept them to implement missing functionality. The rdtime instruction is a notable example.

It’s not that it’s illegal to delegate such exceptions—it just goes against the intended use case. We want Spike to closely mirror the behavior of actual HW implementations, so flexibility beyond what is likely to be implemented is of negative utility.

msuozzo, 2018-04-20:
Makes sense. I don't think I appreciated M-mode's role as a sort of ISA virtualization level. Thanks!

mehnadnerd, 2022-07-26:
https://github.com/riscv-software-src/riscv-isa-sim/pull/1050 support this.

closed 2022-07-26 by aswaterman