riscv-software-src/riscv-isa-sim #195: How to print information without pk?

YohnWang, 2018-04-23:
I want to transplant ucosii on spike,but how to print information without pk?
I refered to the transplantation of freeRTOS (https://github.com/illustris/FreeRTOS-RISCV).
It used 'tohost' to implement printf.
But I don't know how it works.

noureddine-as, 2018-05-07:
Hello,

Check https://github.com/riscv/riscv-tests/blob/master/benchmarks/common/syscalls.c
it contains definition of some functions using the HTIF interface. You can then use that to build your baremetal applications.

My understanding is that tohost and fromhost are memories used for communication between the tethered RISC-V systems and the front end server riscv-fesvr.

Best Regards.

YohnWang, 2018-05-08:
Your link is really helpful,thank you.

tohost and fromhost maybe work by symbols.

jerryz123, 2024-01-08:
That comment is out-of-date. There are no plans for removing HTIF (but there are also no plans for improving it).

> An alternative is clearly needed.
I'm not really sure what you are trying to do.

ekliptik, 2024-01-09:
> I'm not really sure what you are trying to do.

As a dev for an embedded toolchain vendor, I would like to provide simple "hello world" support code to the customer that is guaranteed to be correct to be safe. Using pk isn't "simple" so we have code that lets newlib/picolib directly use HTIF for stdio. Currently, we use an [HTIF implementation from lowRISC](https://github.com/lowRISC/epic-c-example/blob/main/htif.c). This function is broken when it's inlined into a putchar function, it causes spike to print zero bytes. I am hotfixing it with `__attribute__(noinline)`, but I can't find the source of the problem. I might find a compiler bug if I take another look at the assembly but I can't help but think that if there's no spec, then I don't know what conditions have to be met for HTIF to work, so there's no bug. This is a problem. For example, what if it works uninlined just by accident, and it gets miscompiled with a different LLVM version?

I looked into using UART instead. I could rewrite the support code using it instead of HTIF, but I would have to advise the toolchain user to edit and compile the device tree sources for spike, since UART is off by default. Therefore for now I have decided to stick with HTIF and the attribute-based hotfix.

ekliptik, 2024-01-11:
I have found a more proper solution in [riscv-coremark](https://github.com/riscv-boom/riscv-coremark/blob/master/riscv64-baremetal/syscalls.c). The lowRISC code was missing a pair of `__sync_synchronize();` barriers. The noinline attribute is no longer necessary

jerryz123, 2024-01-11:
You may also be interested in libgloss. 

https://github.com/ucb-bar/libgloss-htif