riscv-software-src/riscv-isa-sim #200: dasm without pseudo/alias instrucitons

neelgala, 2018-05-01:
Hi,
Not sure if this has been asked somewhere else. If so, please point me to it.

I am working on building a python-based trace analyser taking the log file generated from "spike -l" as the input. The tool will basically spit out stats like dependencies, instruction histogram, etc. 

Currently spike-dasm dumps the instructions in their pseudo/alias form. 
Eg. li instead of addi. 

Question: is there a one-to-one mapping of pseudo instructions to base-instructions for spike-dasm? if not is it possible to change the dasm to spit out base-instructions instead of pseudo ops.

objdump enables this by the "-M" switch followed by no-aliases. is there something similar in spike-dasm to do the same?

neelgala, 2018-05-01:
Thanks.

I am assuming once the "trace debug" task group finalises on a trace output format, spike would also support that?

I'll probably make a few temporary hacks in my local copy to move forward for now.

Closing the issue

closed 2018-05-01 by neelgala

aswaterman, 2018-05-01:
That seems reasonable to me, but of course that format wouldn't have any disassembled instructions in it.