riscv-software-src/riscv-isa-sim #201: spike --isa=rv32im tries to use the 64bit version of `pk`.  No termination of 32 bit programs

polypolen, 2018-05-05:
hello,

I have a gcc toolchain built for RV32IM.  I experienced the "no termination problem" (e.g. as reported here https://github.com/riscv/riscv-pk/issues/92) and investigated a bit. 

**A possible explanation for this issue is that Spike looks for the 64bit version of `pk` even though we run a 32-bit executable.**   

Spike is configured/built with ` --prefix=$RISCV --with-fesvr=$RISCV --with-isa=rv32im`

The gcc toolchain is configured/built with `--prefix=$RISCV --with-arch=rv32im --with-abi=ilp32  --disable-linux`

PK is configured/built with `--prefix=$RISCV --host=riscv32-unknown-elf --enable-32bit`.

Here I suspect that Spike finds `pk` in the current directory, built for 32bit, and everything is fine:
```
$ cd $TOP/riscv-pk/build
$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c && riscv32-unknown-elf-gcc -o hello hello.c && spike pk hello
Hello world!
```

If I run Spike from another directory, spike fallbacks to the 64bit version of `pk`:
```
damien@gre051335:~/projets/SERENE/src/polen-infra/riscv-tools(build-rv32im)          sam. mai 05 11:21:19
$ cd $TOP
$ find -name pk
./riscv-pk/pk
./riscv-pk/build/pk
./riscv/riscv64-unknown-elf/bin/pk
./riscv/riscv32-unknown-elf/bin/pk
$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c && riscv32-unknown-elf-gcc -o hello hello.c && spike pk hello
    # here the program does not terminate. Halted with ctrl-C.
^C: 
core   0: 0x0000000080000004 (0x34011173) csrrw   sp, mscratch, sp
: 
core   0: 0xffffffff80000008 (0x1a010063) beqz    sp, pc + 416
: 
core   0: 0xffffffff8000000c (0x04a13823) sd      a0, 80(sp)
core   0: exception trap_illegal_instruction, epc 0xffffffff8000000c
core   0:           tval 0x0000000000000000
: 
core   0: 0x0000000080000004 (0x34011173) csrrw   sp, mscratch, sp
```

Renaming the 64bit version of `pk` makes Spike unhappy:
```
$ cd $TOP
$ find -name pk
./riscv-pk/pk
./riscv-pk/build/pk
./riscv/riscv64-unknown-elf/bin/pk
./riscv/riscv32-unknown-elf/bin/pk
$ mv ./riscv/riscv64-unknown-elf/bin/pk ./riscv/riscv64-unknown-elf/bin/pk64 
$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c && riscv32-unknown-elf-gcc -o hello hello.c && spike pk hello
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open pk (did you misspell it? If VCS, did you forget +permissive/+permissive-off?)
```

Last, copying the 32-bit version of `pk` in the current directory makes Spike happy again:
```
$ cd $TOP
$ mv ./riscv/riscv64-unknown-elf/bin/pk64 ./riscv/riscv64-unknown-elf/bin/pk
$ find -name pk
./riscv-pk/pk
./riscv-pk/build/pk
./riscv/riscv64-unknown-elf/bin/pk
./riscv/riscv32-unknown-elf/bin/pk
$ cp ./riscv/riscv32-unknown-elf/bin/pk .
$ echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c && riscv32-unknown-elf-gcc -o hello hello.c && spike pk hello
Hello world!
```

emelcher, 2018-05-07:
[This issue](https://github.com/riscv/riscv-pk/issues/39)  is related.

polypolen, 2018-05-08:
https://github.com/riscv/riscv-pk/issues/39

It looks like `spike` does not find `pk` at its expected location. So I assume the problem is more on `spike`'s side.  

Any hints?

nmeum, 2019-03-12:
I think I had the same issue you described, the thing is that spike uses the `RV64IMAFDC` ISA by default. If I attempt to invoke a hello word program cross compiled with `riscv32-unknown-elf-gcc` with this ISA spike doesn't terminate on my system either.

I managed to fix this by invoking spike as `spike --isa=RV32IMAFDC pk test` where test is a hello world program cross compiled with `riscv32-unknown-elf-gcc`.

ibndias, 2020-05-13:
`spike --isa=RV32IMAFDC pk test` works, but the `--isa=RV32IMAFD` doesn't!

I recheck my `riscv-gnu-toolchain`, it was compiled with `--with-arch=rv32gc` as mentioned by the `readme.md`
while my `spike` was compiled with `--with-isa=rv32ima` because i was using `riscv-tools`'s `build-rv32ima.sh`

So the `exception trap_illegal_instruction` was because **the program has compressed instruction that's not supported by the `spike`**