riscv-software-src/riscv-isa-sim #206: Spike --dump-dts shows no DT string - but the help screen

cgarlati, 2018-05-15:
spike --dump-dts shows no DT info but the help screen.

Is this the expected behavior or am I doing something wrong?

$ spike --dump-dts
usage: spike [host options] <target program> [target options]
Host Options:
  -p<n>                 Simulate <n> processors [default 1]
  -m<n>                 Provide <n> MiB of target memory [default 2048]
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
                          at base addresses a and b (with 4 KiB alignment)
  -d                    Interactive debug mode
  -g                    Track histogram of PCs
  -l                    Generate a log of execution
  -h                    Print this help message
  -H                    Start halted, allowing a debugger to connect
  --isa=<name>          RISC-V ISA string [default RV64IMAFDC]
  --pc=<address>        Override ELF entry point
  --hartids=<a,b,...>   Explicitly specify hartids, default is 0,1,...
  --ic=<S>:<W>:<B>      Instantiate a cache model with S sets,
  --dc=<S>:<W>:<B>        W ways, and B-byte blocks (with S and
  --l2=<S>:<W>:<B>        B both powers of 2).
  --extension=<name>    Specify RoCC Extension
  --extlib=<name>       Shared library to load
  --rbb-port=<port>     Listen on <port> for remote bitbang connection
  --dump-dts            Print device tree string and exit
  --progsize=<words>    Progsize for the debug module [default 2]
  --debug-sba=<bits>    Debug bus master supports up to <bits> wide accesses [default 0]
  --debug-auth          Debug module requires debugger to authenticate

seldridge, 2018-05-15:
Sorry about this. It's due to HTIF wanting to always see a target binary, but no binary (correctly) being provided. There's some commentary on this [here](https://github.com/riscv/riscv-fesvr/pull/25#issuecomment-383823880).

A temporary workaround, unless something has changed from my version, is to provide a non-existent binary (this doesn't have to exist):
```
spike --dump-dts foo
```

This is somewhat low-priority on my end, but I'm intending to cleanup the option parsing side of things related to Spike eventually. That will take care of this. [RFC here](https://github.com/riscv/riscv-isa-sim/issues/196).

cgarlati, 2018-05-15:
Thank you Shuyler for the prompt reply. I can confirm the workaround works with the current version of spike - see output below:

```
$ spike --dump-dts anything-goes-here
/dts-v1/;

/ {
  #address-cells = <2>;
  #size-cells = <2>;
  compatible = "ucbbar,spike-bare-dev";
  model = "ucbbar,spike-bare";
  cpus {
    #address-cells = <1>;
    #size-cells = <0>;
    timebase-frequency = <10000000>;
    CPU0: cpu@0 {
      device_type = "cpu";
      reg = <0>;
      status = "okay";
      compatible = "riscv";
      riscv,isa = "rv64imafdc";
      mmu-type = "riscv,sv48";
      clock-frequency = <1000000000>;
      CPU0_intc: interrupt-controller {
        #interrupt-cells = <1>;
        interrupt-controller;
        compatible = "riscv,cpu-intc";
      };
    };
  };
  memory@80000000 {
    device_type = "memory";
    reg = <0x0 0x80000000 0x0 0x80000000>;
  };
  soc {
    #address-cells = <2>;
    #size-cells = <2>;
    compatible = "ucbbar,spike-bare-soc", "simple-bus";
    ranges;
    clint@2000000 {
      compatible = "riscv,clint0";
      interrupts-extended = <&CPU0_intc 3 &CPU0_intc 7 >;
      reg = <0x0 0x2000000 0x0 0xc0000>;
    };
  };
  htif {
    compatible = "ucb,htif0";
  };
};

```
One more question: The help screen shows only a "Host Options" section while the usage string shows optional "target options". Are there any "target options"? How can I display them?
```
usage: spike [host options] <target program> [target options]
```

seldridge, 2018-05-15:
Target options are, somewhat confusingly in the current help text, just the options passed to your binary. If your binary takes no options, then you wouldn't need to provide any.

To clarify, option parsing is currently a multi-level procedure due to how these tools are put together, specifically, `spike` and the Rocket Chip emulator (both Verilator and VCS).

The tool (`spike` here) has some options. It peels off the options it understands and passes the rest on to some classes in libfesvr.so (debug target module/DTM, host target interface/HTIF, etc.) which peels off more options. These are "host options". You then provide a binary and that may, optionally, consume additional options. Those are "target options". Existing help text describes `spike` options as host options.

From the WIP PR that I referenced, the help text becomes clearer (just use this for a rough reference, don't use that PR):
```
> ./build/spike -h
Usage: spike [SPIKE OPTION]... [HOST OPTION]... BINARY [TARGET OPTION]...
Run a BINARY on Spike (the RISC-V ISA Simulator).

Mandatory arguments to long options are mandatory for short options too.

SPIKE OPTIONS
  -p N                     Simulate N processors (default: 1)
  -m N                     Provide N MiB of target memory (default: 2048)
  -m A:M,B:N,...           Provide memory regions of size M and N bytes at base
                             address A and B (with 4KiB alignment)
  -d                       Interactive debug mode
  -g                       Track historgram of PCs
  -l, --verbose            Generate a log of execution
  -h, --help               Print this help message and exit
  -H, --halted             Start halted, allowing a debugger to connect
      --isa=NAME           RISC-V ISA string to use (default: RV64IMAFDC)
      --pc=ADDRESS         Override ELF entry point
      --hartids=A,B,...    Explicitly specify hartids, default is 0,1,...
      --ic=S:W:B           Instantiate an instruction cache with S sets, W ways,
                             and B-byte blocks (with S and B both powers of 2)
      --dc=S:W:B           Instantiate a data cache with S sets, W ways,
                             and B-byte blocks (with S and B both powers of 2)
      --l2=S:W:B           Instantiate an L2 cache with S sets, W ways,
                             and B-byte blocks (with S and B both powers of 2)
      --extension=NAME     Specify RoCC extension NAME to use
      --extlib=NAME        Load shared library NAME
      --rbb-port=PORT      Listen on PORT for remote bitbang connection
      --dump-dts           Pirnt device tree string and exit
      --progsize=WORDS     Set program size for the debug module (default: 2)
      --debug-sba=BITS     Set debug bus master to support up to BITS wide
                             accesses (default: 0)
      --debug-auth         Set debvg module to require authentication

HOST OPTIONS
  -h, --help               Display this help and exit
       +permissive         The host will ignore any unparsed options up until
                             +permissive-off (Only needed for VCS)
       +permissive-off     Stop ignoring options. This is mandatory if using
                             +permissive (Only needed for VCS)
      --rfb=DISPLAY        Add new remote frame buffer on display DISPLAY
       +rfb=DISPLAY          to be accessible on 5900 + DISPLAY (default = 0)
      --signature=FILE     Write torture test signature to FILE
       +signature=FILE
      --chroot=PATH        Use PATH as location of syscall-servicing binaries
       +chroot=PATH

HOST OPTIONS (currently unsupported)
      --disk=DISK          Add DISK device. Use a ramdisk since this isn't
       +disk=DISK            supported

TARGET (RISC-V BINARY) OPTIONS
  These are the options passed to the program executing on the emulated RISC-V
  microprocessor.
```

For comparison, take a look at the help text of the Rocket Chip emulator. Host options are shared, while emulator options are different from `spike`:
```
> ./emulator-freechips.rocketchip.system-DefaultConfig -h
Usage: ./emulator-freechips.rocketchip.system-DefaultConfig [EMULATOR OPTION]... [VERILOG PLUSARG]... [HOST OPTION]... BINARY [TARGET OPTION]...
Run a BINARY on the Rocket Chip emulator.

[... 31 lines cut ...]
       +permissive-off     Stop ignoring options. This is mandatory if using
                             +permissive (Only needed for VCS)
      --rfb=DISPLAY        Add new remote frame buffer on display DISPLAY
       +rfb=DISPLAY          to be accessible on 5900 + DISPLAY (default = 0)
      --signature=FILE     Write torture test signature to FILE
       +signature=FILE
      --chroot=PATH        Use PATH as location of syscall-servicing binaries
       +chroot=PATH

HOST OPTIONS (currently unsupported)
      --disk=DISK          Add DISK device. Use a ramdisk since this isn't
       +disk=DISK            supported

TARGET (RISC-V BINARY) OPTIONS
  These are the options passed to the program executing on the emulated RISC-V
  microprocessor.

EXAMPLES
  - run a bare metal test:
    ./emulator-freechips.rocketchip.system-DefaultConfig $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
  - run a bare metal test showing cycle-by-cycle information:
    ./emulator-freechips.rocketchip.system-DefaultConfig +verbose $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add 2>&1 | spike-dasm
  - run an ELF (you wrote, called 'hello') using the proxy kernel:
    ./emulator-freechips.rocketchip.system-DefaultConfig pk hello
```