riscv-software-src/riscv-isa-sim #211: Dynamic instruction count in spike? (updated)

AhmedSamara, 2018-06-11:
I saw issue #14 but this appears to be deprecated, when I run this I just get:

```
spike pk -s a.out 
../machine/mtrap.c:18: machine mode: unhandlable trap 4 @ 0x000000008000482e
```

aswaterman, 2018-06-11:
That looks like an issue unrelated to the -s flag (but it's hard to say what it is).

AhmedSamara, 2018-06-11:
Everything works fine without a -s flag.

aswaterman, 2018-06-11:
I can't replicate it.  Can you upload your pk and a.out binaries somewhere?

aswaterman, 2018-06-11:
and also indicate which commit of pk you are using?

AhmedSamara, 2018-06-11:
https://ufile.io/pckjp

I am using [riscv-pk commit 3d921d3c76db3af7b9ae0b5df0f0790f26222246](https://github.com/riscv/riscv-pk/tree/3d921d3c76db3af7b9ae0b5df0f0790f26222246)

aswaterman, 2018-06-11:
Looks like it's getting miscompiled.  https://github.com/riscv/riscv-pk/blob/3d921d3c76db3af7b9ae0b5df0f0790f26222246/machine/unprivileged_memory.h#L70 is getting compiled instead of the corresponding `#else` for some reason.

I don't have time to help debug why that might be, but it should at least give you a starting point.

AhmedSamara, 2018-06-11:
> riscv-pk/blob/3d921d3c76db3af7b9ae0b5df0f0790f26222246/machine/
> unprivileged_memory.h#L70 is getting compiled instead of the

Thank you for your help.

closed 2018-06-13 by AhmedSamara

AhmedSamara, 2018-06-18:
Question: why is logging dependant on compressed instructions being added?

aswaterman, 2018-06-18:
Because the rdcycle instruction causes an illegal-instruction trap, and the illegal-instruction trap handler needs to know whether 32-bit-wide instructions are always naturally aligned.

AhmedSamara, 2018-06-19:
So would this also fail under the 64 bit version of riscv-tools?

aswaterman, 2018-06-19:
Possibly, but whether it actually fails depends on the alignment of the rdcycle instruction.