riscv-software-src/riscv-isa-sim #215: clearing load_reservation on xret

pmundkur, 2018-07-09:
The load reservation is set on virtual addresses, but xRET doesn't yield the reservation as is done on traps.  
Isn't this a bug?

aswaterman, 2018-07-09:
I forgot to update Spike with respect to this change: https://github.com/riscv/riscv-isa-manual/blob/master/src/a.tex#L213

Will do soon.

pmundkur, 2018-07-09:
Ok, but could you confirm whether not yielding on xret is a bug?

aswaterman, 2018-07-09:
Since the load reservation is on virtual addresses, I think the correct thing to do is to yield the reservation on SFENCE.VMA.

Yielding on xRET is not required (but is certainly allowed).

pmundkur, 2018-07-09:
Ok, makes sense, thanks.  If you'd like, I can submit a fix right now, since I'm poking around there.

aswaterman, 2018-07-09:
Thanks for the offer, but I'd like to think about it some more first.  I need to convince myself that using virtual addresses is actually valid, or if we should just change Spike to reserve physical addresses.

pmundkur, 2018-07-09:
Got it, sounds good.

aswaterman, 2018-07-09:
I've concluded that it's only valid to use virtual addresses if there are no cases that a store-conditional could succeed in this scheme that would have failed using physical addresses.  The troublesome case is due to TLB incoherence: i.e. when software forgets to use SFENCE.VMA.

This could be fixed by yielding the reservation on TLB misses, but I figure the cleanest thing to do is to start using physical addresses instead.  Patch incoming soon.

Thanks for pointing this out!

closed 2018-07-10 by aswaterman

ccelio, 2018-07-10:
Sorry I'm a bit late (and not to dissuade you from using PAs), but you moved to using physical addresses just in case software forgets to use an SFENCE.VMA? Isn't software already screwed  for other reasons in that scenario? Sorry I just want to make sure I fully understand.

aswaterman, 2018-07-10:
Sort of, but not exactly.  I claim that using VAs is incorrect.  The only time the incorrect behavior can manifest is when the software is incorrect, too.  But that doesn't make it OK for the HW/Spike to be wrong.

I think using VAs could've been made fully correct by yielding the reservation on DTLB misses, but didn't want to prove & explain it.  Switching to PAs was the lazy solution.

PR #217 (2018-07-10): Refactor and fix LR/SC implementation
- Use physical addresses to avoid homonym ambiguity (closes #215)

- Disallow LR/SC to non-memory-like regions

- Yield reservation on store-conditional (https://github.com/riscv/riscv-isa-manual/commit/03a5e722fc0fe7b94dd0a49f550ff7b41a63f612)

- Don't yield reservation on exceptions (it's no longer required).
commit 95487c24 (2018-07-10): Refactor and fix LR/SC implementation (#217)

- Use physical addresses to avoid homonym ambiguity (closes #215)

- Yield reservation on store-conditional (https://github.com/riscv/riscv-isa-manual/commit/03a5e722fc0fe7b94dd0a49f550ff7b41a63f612)

- Don't yield reservation on exceptions (it's no longer required).