riscv-software-src/riscv-isa-sim #219: Read ustatus creates an illegal instruction execution

pierregobin, 2018-07-18:
The sample code here below makes an illegal instruction at execution when reading the csr 'ustatus' 

`

> include <stdio.h>
> 
> int main(int argc, char *argv[])
> {
>         int ustatus;
>         asm("csrr %0,ustatus" :"=r"(ustatus)::);
>         printf("ustatus=%d\n",ustatus);
> }
> 

`

The compilation command is 

> riscv32-unknown-elf-gcc  -march=rv32im -mabi=ilp32 -o readcsr.out -O1 readcsr.c

The execution (`spike pk readcsr.out`) yields the following message 

 `z  00000000 ra 000100b0 sp 7fbe5d90 gp 00023600
tp 00000000 t0 00010374 t1 00000000 t2 00000000
s0 00000000 s1 00000000 a0 00000001 a1 7fbe5da4
a2 00000000 a3 00000000 a4 00000001 a5 00000002
a6 0000001f a7 00000000 s2 00000000 s3 00000000
s4 00000000 s5 00000000 s6 00000000 s7 00000000
s8 00000000 s9 00000000 sA 00000000 sB 00000000
t3 00000000 t4 00000000 t5 00000000 t6 00000000
pc 000101b0 va 000025f3 insn       000025f3 sr 00040020
An illegal instruction was executed!
`

And the instruction that triggers this exception is ;
 101b0:	000025f3          	csrr	a1,ustatus

Is it a normal behaviour of the rv ? or did I make a mistake somewhere ?

aswaterman, 2018-07-18:
Spike doesn't implement the N extension for user-mode interrupts, so trapping on ustatus access is the expected behavior.

pierregobin, 2018-07-18:
Thanks for your prompt answer - I agree (I re-read riscv-privilege-v1.10).

Initially I was trying to read mstatus instead of ustatus - and I got already the exception as shown below (you can reproduce this by changing ustatus by mstatus within the asm statement).

> pc 000101b0 va 300025f3 insn       300025f3 sr 00040020
> An illegal instruction was executed!

I was suspecting that the exception was triggered due to mode permission, this is why I choosed to read ustatus  (which can be read from Umode onward) instead of mstatus (readable only by Mmode). I was thinking that the startup file was moving the mode of riscv from machine mode to user mode, and that could have been the explanation of the exception when accessing mstatus, but if I am not mistaken, main() is executed in machine mode. 

For the time being, I do not understand why the error occurs with mstatus. Would you have any idea ?
Do you know what 'sr' means for in the dump frame here above ?

aswaterman, 2018-07-18:
The program executes in U-mode, so it’s illegal to access the more-privileged mstatus CSR.

What are you aiming to accomplish?

pierregobin, 2018-07-18:
I am trying to play with interrupts on RISCV - and before making verilog simulation, I wanted to check some basic code running on spike.
As a first target I would like to trigger software interrupt - and prior to this I was checking the access to some CSR.

gsauthof, 2020-01-18:
I ran into this issue as well.

What I'm trying to accomplish is to detect in a user-mode program if the vector extension is enabled or not. Based on the result the program is then able to fall-back to a non-vector implementation or exit  with a nice error message/code.

I understand that any access (including read-only) to `mstatus`/`sstatus` raises illegal instruction exceptions because the user privilege level is too low.

I assume that the VS vector status is also mirrored in `ustatus[24:23]` thus being able to access that would solve my use case.

Without read-only access to the VS from user-mode I can only think of this detection approach: execute a dummy `vsetvli` instruction and somehow catch the illegal instruction exception and continue execution.

But I don't know how to do that exactly and whether this would work with Spike.

There is also the `misa` status register which looks promising, but it doesn't seem to be mirrored into another register that is accessible in user-mode (such as `uisa` or something like that).