riscv-software-src/riscv-isa-sim #223: Debugging a program through OpenOCD

Chakri-myneni, 2018-08-02:
I am trying to debug a program through openOCD following the steps from [https://github.com/riscv/riscv-isa-sim](url)

Files used for debugging are rot13.c, spike.cfg from the above mentioned link.
After compiling,
-> For **spike --rbb-port=9824 -m0x10000000:0x20000 rot13-64** this command, it is listening on port 9824.
[image]

-> For **openocd -d -f spike.cfg** this command the result is as follows
[image]

Tool chain is latest.(i.e updated and built on 2nd Aug 2018)

My questions are 
1. what can be the reason for the error ?
2. what are the necessary changes to be made?
3. Do I need to enable something while building the toolchain? If so what should be enabled?

timsifive, 2018-08-02:
See discussion at https://forums.sifive.com/t/debugging-a-program-through-openocd/1471

Chakri-myneni, 2018-08-03:
@timsifive that question in sifive forums is also posted by me (Chakri).

benjschro, 2018-08-03:
For what it is worth, I am running into this EXACT same problem.  For the whole toolchain and additional tools, I cloned them from the git repos a couple days ago.  I am running this all in ubuntu-18.04.1-live-server-amd64 in a virtual machine.

Chakri-myneni, 2018-08-07:
Thank you @timsifive  and @benjschro for your help.

Reason for above problem is my Ubuntu version on which toolchain is installed. So I installed toolchain on Ubuntu 18.04 and OpenOCD  is working.
Chakri

closed 2018-08-07 by Chakri-myneni