riscv-software-src/riscv-isa-sim #224: MMU seg fault (Booting Linux)

suedadam, 2018-08-07:
I'm trying to boot a standard version of Linux (from https://github.com/riscv/riscv-linux) and I'm hitting a problem when it gets to setting up the MMU (https://github.com/riscv/riscv-linux/blob/master/arch/riscv/kernel/head.S#L96)

Log:
```
: r 30
core   0: 0x00000000800120a6 (0x028000ef) jal     pc + 0x28
core   0: 0x00000000800120ce (0x800005b7) lui     a1, 0x80000
core   0: 0x00000000800120d2 (0x00000517) auipc   a0, 0x0
core   0: 0x00000000800120d6 (0xfa650513) addi    a0, a0, -90
core   0: 0x00000000800120da (0x00008d89) sub     a1, a1, a0
core   0: 0x00000000800120dc (0x000090ae) add     ra, ra, a1
core   0: 0x00000000800120de (0x00000517) auipc   a0, 0x0
core   0: 0x00000000800120e2 (0x03650513) addi    a0, a0, 54
core   0: 0x00000000800120e6 (0x0000952e) add     a0, a0, a1
core   0: 0x00000000800120e8 (0x10551073) csrw    stvec, a0
core   0: 0x00000000800120ec (0x00772617) auipc   a2, 0x772
core   0: 0x00000000800120f0 (0xf8c60613) addi    a2, a2, -116
core   0: 0x00000000800120f4 (0x00008231) srli    a2, a2, 12
core   0: 0x00000000800120f6 (0xfff0059b) addiw   a1, zero, -1
core   0: 0x00000000800120fa (0x03f59593) slli    a1, a1, 63
core   0: 0x00000000800120fe (0x00008e4d) or      a2, a2, a1
core   0: 0x0000000080012100 (0x0001e517) auipc   a0, 0x1e
core   0: 0x0000000080012104 (0xf7850513) addi    a0, a0, -136
core   0: 0x0000000080012108 (0x00008131) srli    a0, a0, 12
core   0: 0x000000008001210a (0x00008d4d) or      a0, a0, a1
core   0: 0x000000008001210c (0x12000073) sfence.vma (args unknown)
core   0: 0x0000000080012110 (0x18051073) csrw    satp, a0
core   0: exception trap_instruction_page_fault, epc 0x0000000080012114
core   0:           tval 0x0000000080012114
core   0: exception trap_instruction_page_fault, epc 0xffffffff8000009c
core   0:           tval 0xffffffff8000009c
core   0: exception trap_instruction_page_fault, epc 0xffffffff8000009c
core   0:           tval 0xffffffff8000009c
core   0: exception trap_instruction_page_fault, epc 0xffffffff8000009c
core   0:           tval 0xffffffff8000009c

```

Command run:
```
spike -d ./bbl
```

I'm not really sure how to debug this further since the `mem` command only supports physical addresses

aswaterman, 2018-08-07:
The first trap is expected (as the comment at https://github.com/riscv/riscv-linux/blob/master/arch/riscv/kernel/head.S#L88 suggests).   The subsequent trap is most definitely not expected.

@palmer-dabbelt is there any possibility that MAXPHYSMEM_2GB got broken on RV64?

Can you try enabling MAXPHYSMEM_128GB and CMODEL_MEDANY and see if it fails similarly?

suedadam, 2018-08-07:
@aswaterman Same thing with 128GB mode and MEDANY.

aswaterman, 2018-08-07:
Thanks.  This is past my ability to debug over github. It smells like the kernel isn't setting up the page tables correctly.

FWIW, the **mem** command does support virtual memory, if it's enabled on the core (i.e., after the CSRW to satp).  You can then do something like **mem 0 ffffffff8000009c**; the 0 indicates which core's address translation is to be used.

suedadam, 2018-08-07:
For the 128GB max memory version, here's the output when I ran `mem` against the virtual address:

```
: mem 0 0xffffffe0000000a0
: 

```
(Which I've deduced means it's not a valid address since it doesn't respond at all). 

Here's the log if it helps at all (I'm pretty stuck myself...):

```
: r 30
core   0: 0x00000000800120a6 (0x028000ef) jal     pc + 0x28
core   0: 0x00000000800120ce (0xfff0059b) addiw   a1, zero, -1
core   0: 0x00000000800120d2 (0x02559593) slli    a1, a1, 37
core   0: 0x00000000800120d6 (0x00000517) auipc   a0, 0x0
core   0: 0x00000000800120da (0xfa250513) addi    a0, a0, -94
core   0: 0x00000000800120de (0x00008d89) sub     a1, a1, a0
core   0: 0x00000000800120e0 (0x000090ae) add     ra, ra, a1
core   0: 0x00000000800120e2 (0x00000517) auipc   a0, 0x0
core   0: 0x00000000800120e6 (0x03650513) addi    a0, a0, 54
core   0: 0x00000000800120ea (0x0000952e) add     a0, a0, a1
core   0: 0x00000000800120ec (0x10551073) csrw    stvec, a0
core   0: 0x00000000800120f0 (0x007fa617) auipc   a2, 0x7fa
core   0: 0x00000000800120f4 (0xf8860613) addi    a2, a2, -120
core   0: 0x00000000800120f8 (0x00008231) srli    a2, a2, 12
core   0: 0x00000000800120fa (0xfff0059b) addiw   a1, zero, -1
core   0: 0x00000000800120fe (0x03f59593) slli    a1, a1, 63
core   0: 0x0000000080012102 (0x00008e4d) or      a2, a2, a1
core   0: 0x0000000080012104 (0x0001f517) auipc   a0, 0x1f
core   0: 0x0000000080012108 (0xf7450513) addi    a0, a0, -140
core   0: 0x000000008001210c (0x00008131) srli    a0, a0, 12
core   0: 0x000000008001210e (0x00008d4d) or      a0, a0, a1
core   0: 0x0000000080012110 (0x12000073) sfence.vma (args unknown)
core   0: 0x0000000080012114 (0x18051073) csrw    satp, a0
core   0: exception trap_instruction_page_fault, epc 0x0000000080012118
core   0:           tval 0x0000000080012118
core   0: exception trap_instruction_page_fault, epc 0xffffffe0000000a0
core   0:           tval 0xffffffe0000000a0
core   0: exception trap_instruction_page_fault, epc 0xffffffe0000000a0
core   0:           tval 0xffffffe0000000a0
core   0: exception trap_instruction_page_fault, epc 0xffffffe0000000a0
core   0:           tval 0xffffffe0000000a0
core   0: exception trap_instruction_page_fault, epc 0xffffffe0000000a0
core   0:           tval 0xffffffe0000000a0
core   0: exception trap_instruction_page_fault, epc 0xffffffe0000000a0
core   0:           tval 0xffffffe0000000a0
core   0: exception trap_instruction_page_fault, epc 0xffffffe0000000a0
core   0:           tval 0xffffffe0000000a0
: mem 0 0xffffffe0000000a0
: 

```

aswaterman, 2018-08-07:
If it prints nothing, then that address isn't mapped.

palmer-dabbelt, 2018-08-07:
The best bet for a compatible set of versions is what's in freedom-u-sdk.  You should be able to "make sim" there and get a spike run of Linux, if that doesn't work then it's a real bug.  We're in the process of pushing through a big code review to get the interrupt patches upstream in Linux so the various branches are probably unstable right now, so if you're trying to work directly upstream your best bet is to for a riscv-linux-4.18 which should be coming soon (4.18 was delayed a week, otherwise it'd be out now :)).

Sorry things are a bit of a mess in Linux land, but my hope is that we'll have the base interrupt code in 4.18!

suedadam, 2018-08-07:
@palmer-dabbelt Got it working with the `freedom-u-sdk` submodule's version. Thanks! I'm facing another issue with it but I'll make another issue task for it!

closed 2018-08-07 by suedadam