riscv-software-src/riscv-isa-sim #226: spike error: cannot open linker script file spike.lds: No such file or directory

sharinoz, 2018-08-07:
hi
when I try to compile my c code I get this error:

rotemshahar@asic2-serv3:~$ ./rocket-chip/riscv-tools/bin/riscv64-unknown-elf-gcc -g -Og -o hello-64.o -c hello.c
rotemshahar@asic2-serv3:~$ ./rocket-chip/riscv-tools/bin/riscv64-unknown-elf-gcc -g -Og -T spike.lds -nostartfiles -o hello-64 hello-64.o
/home/rotemshahar/rocket-chip/riscv-tools/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: cannot open linker script file spike.lds: No such file or directory
collect2: error: ld returned 1 exit status

I have no idea why this is happening, I have installed the spike as explained in the Code section
please help me to make it work

thanks
Shahar

aswaterman, 2018-08-07:
This error actually has nothing to do with Spike. GCC is emitting an error when attempting to link some program using a linker script that happens to be named spike.lds.

sharinoz, 2018-08-08:
Ok, so it's a gcc issue. How can I fix it?

Thanks