riscv-software-src/riscv-isa-sim #228: spike-dasm issue

Chakri-myneni, 2018-08-09:
Hello everyone,
When I run **cat spike.log | spike-dasm**  the output is exactly same as spike.log. Which means my log file is directly getting printed on screen without going to spike-dasm. I even tried **cat spike.log | $RISCV/bin/spike-dasm**, but the result is same as spike.log.

Here is the picture of my log
[image]

My question
1. how do I get disassembled output(using spike-dasm) of my log file that is generated through **spike -l** ?

Thank you,
Chakri

neelgala, 2018-08-09:
spike-dasm will disassemble the contents within the DASM(hexvalue_here) string. Example:

echo "DASM(0x00000297)"| spike-dasm 

will print 

auipc   t0, 0x0

Since the spike.log doesn't have the DASM directive spike-dasm will not print anything extra.

Chakri-myneni, 2018-08-09:
Thank you @neelgala 

An output like following pic
[image]

Is it possible using Rocket only or can be done using spike also?

closed 2020-11-27 by Chakri-myneni