riscv-software-src/riscv-isa-sim #230: User fetch segfault while running spike -d pk

sharinoz, 2018-08-14:
hi
I wrote this simple code:

#include <stdio.h>
 int main(void) { 
	volatile int i=3;
	printf ("%d\n",i);
	//printf("Hello world!\n"); 
/* 	volatile int i=0;
	while(i<3){
		printf ("%d\n",i);
		i++;
	} */
	return 0; 
}

then i did this:

rotemshahar@asic2-serv3:~/riscv-isa-sim$ riscv64-unknown-elf-gcc -g -Og -o hello-64.o -c hello.c
rotemshahar@asic2-serv3:~/riscv-isa-sim$ spike -d pk hello-64.o 2>hello_log

and the result is:

z  0000000000000000 ra 0000000000000000 sp 000000007f7e9b50 gp 0000000000000000
tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
s0 0000000000000000 s1 0000000000000000 a0 0000000000000000 a1 0000000000000000
a2 0000000000000000 a3 0000000000000000 a4 0000000000000000 a5 0000000000000000
a6 0000000000000000 a7 0000000000000000 s2 0000000000000000 s3 0000000000000000
s4 0000000000000000 s5 0000000000000000 s6 0000000000000000 s7 0000000000000000
s8 0000000000000000 s9 0000000000000000 sA 0000000000000000 sB 0000000000000000
t3 0000000000000000 t4 0000000000000000 t5 0000000000000000 t6 0000000000000000
pc 0000000000000000 va 0000000000000000 insn       ffffffff sr 8000000200046020
User fetch segfault @ 0x0000000000000000

i have updated my folder (did git submodule update --init --recursive and then configure and make make install) just today
I have tried to run another simple code that came into the exact same result (User fetch segfault):
#include <stdio.h>
 int main(void) {
        volatile int a=3;
        volatile int b=2;
        volatile int sum=0;
        sum=a+b;
        return 0;
}
i also tried without the volatile - it got me the same result

please help me understand what am I doing wrong
thanks
Shahar

singhalshubh, 2018-12-16:
Hey shahar
Try using this command : spike -d pk hello-64 2>hello_log
My terminal screenshot : 
shubh@shubh-HP-Pavilion-Notebook:~/Documents$ riscv64-unknown-elf-gcc issue.c -o issue
shubh@shubh-HP-Pavilion-Notebook:~/Documents$ spike -d /scratch/tools/riscv/riscv64-unknown-elf/bin/pk issue 2>issue_log
3
shubh@shubh-HP-Pavilion-Notebook:~/Documents$ spike -d /scratch/tools/riscv/riscv64-unknown-elf/bin/pk issue.o 2>issue_log
couldn't open ELF program: issue.o!
shubh@shubh-HP-Pavilion-Notebook:~/Documents$ 
It works...

emc2314, 2019-03-21:
Use -static when compiling

erlingrj, 2019-09-25:
I am experiencing similar problems and need guidance. I am trying to simulate a very simple program written in assembly: basic.s
```
.section .text
.globl _start
_start:
    li t3, 25 
    li t4, 15
    add t2, t3, t4
    beq t2, zero, b1
    j b2

b1:
    sub t1, t4, t3
    j end

b2:
    add t1, t4, t3

end:

``` 

Compile
**> $RISCV/bin/riscv64-unknown-elf-as basic.s -o basic**

Print object dump
**> $RISCV/bin/riscv64-unknown-elf-objdump -d basic**

> basic:     file format elf64-littleriscv
> 
> 
> Disassembly of section .text:
> 
> 0000000000000000 <_start>:
>    0:	01900e13          	li	t3,25
>    4:	00f00e93          	li	t4,15
>    8:	01de03b3          	add	t2,t3,t4
>    c:	00038463          	beqz	t2,14 <b1>
>   10:	00c0006f          	j	1c <b2>
> 
> 0000000000000014 <b1>:
>   14:	41ce8333          	sub	t1,t4,t3
>   18:	0080006f          	j	20 <end>
> 
> 000000000000001c <b2>:
>   1c:	01ce8333          	add	t1,t4,t3

Then trying to run it on the spike simulator
**> $RISCV/bin/spike -d pk basic 2>log**

> z  0000000000000000 ra 0000000000000000 sp 000000007f7e9b50 gp 0000000000000000
> tp 0000000000000000 t0 0000000000000000 t1 0000000000000000 t2 0000000000000000
> s0 0000000000000000 s1 [...]

**> cat log**
This prints out many many lines with instructions I dont recognize. But it ends up with these two instructions repeated forever:

> : core   0: 0x0000000080002ca8 (0x0000dffd) c.beqz  a5, pc - 2
> : core   0: 0x0000000080002ca6 (0x0000601c) c.ld    a5, 0(s0)

Thanks for your help

rgdaddio, 2020-10-23:
Add proper end and will exit correctly from spike at least.

end:
	li a7, 93			#exit system call
	ecall