riscv-software-src/riscv-isa-sim #232: Processor.cc : disassembler getting created before max_xlen is set by parse isa

smlardin, 2018-08-21:
It looks like there was a recent change to processor.cc to move the disassembler creation before parse_isa_string. The problem is max_xlen is not set until the parse_isa_string is parsed so the disassembler is created using a max_xlen not associated with the ISA. Was this change intentional? It matters because the disassembler can decode compressed instructions differently (wrongly) based on the max_xlen.

processor_t::processor_t(const char* isa, simif_t* sim, uint32_t id,
        bool halt_on_reset)
  : debug(false), halt_request(false), sim(sim), ext(NULL), id(id),
  halt_on_reset(halt_on_reset), last_pc(1), executions(1)
{
  disassembler = new disassembler_t(max_xlen);
  parse_isa_string(isa);   // this sets max_xlen
  register_base_instructions();

  mmu = new mmu_t(sim, this);

  reset();
}

aswaterman, 2018-08-21:
This is a regression caused by 2cd60b277e909a5599ca48e4561cbfbc61460186, which was fixing a different bug.

aswaterman, 2018-08-21:
I think the fix is to roll back that commit, and fix the underlying bug in a different way, by not registering the extension inside of parse_isa_string.

aswaterman, 2018-08-21:
Should be resolved by b6ec196e9efe33d29d0c9fb80202737719c7730f

closed 2018-08-21 by aswaterman