riscv-software-src/riscv-isa-sim #236: Riscv function simulation model ???

IC-Dream, 2018-09-07:
Hi,

Is there a riscv function simulation model ? we could get the golden information using the model, like pc,instruction executed, updated register.

Thanks
Dream

aswaterman, 2018-09-07:
That's what this repo is.  You can have Spike dump some trace information with the `-l` flag.

IC-Dream, 2018-09-07:
Thanks a lot, I have tried, but i didn't see any execution result in the trace information. for example, the following "addi", the updated value was not printed. 
[image]

aswaterman, 2018-09-07:
You can also check out the commit-log feature (run the configure script with --enable-commit-log).  It might serve your purpose.