riscv-software-src/riscv-isa-sim #238: Difference between interactive debug mode and "normal" mode

lmorel-insa, 2018-09-18:
Dear all, 
I have a rather general question about the internals of spike.

**Context**
I'm modifying spike (riscv32ima) to study some extension I want to propose (I believe the goal of this extension is off-topic for the moment). 
Suffice it to say that i modify how certain instructions are interpreted, hence modifying
- files inside riscv-isa-sim/riscv/insns
- also mmu.h where the code that is dealing with fetching instructions is written. 

My modifications work properly when executing spike in a step-by-step fashion, ie passing the -d option and either stepping through programs one instruction at a time or using the rs (run silent) command. 

However, my modifications trigger a bug (not of spike itself but of my extension) when I execute spike without the -d option. 

**Question**
In a way, I'm trying to figure out what's the difference between -d and not(-d), more specifically, why this bug is triggered without -d and why it is NOT with -d. 

So I guess my question is: how is spike fetching/decoding instructions in its "non-interactive-debug" mode ? How does that differ from what happens when the -d option is used.  I think I understand what's going on in slow_path_mode. For example: When using -d and executing with run silent,  instructions are batch-decoded by the mmu. That is probably the use of access_icache(mmu.h:246, commit bed0a54 on https://github.com/riscv/riscv-isa-sim/blob/master/riscv/mmu.h).

But are there other differences ? 

I'm ready to read whatever one might find useful to send me, but i've been unable to find proper documentation of spike's internals and the documentation projet here https://github.com/poweihuang17/Documentation_Spike 
I'm willing to incorporate into this document whatever I can gather around my question!

aswaterman, 2018-09-18:
With respect to this issue, the main difference between -d and not-d is that not-d uses an instruction cache to speed up execution, and -d bypasses the instruction cache and fetches the instruction from memory each time.  The instruction cache is not coherent (until FENCE.I is executed) so erroneous code can behave differently between the two.  (Although the executions are different, both are valid per the RISC-V spec.)

damaxi, 2018-11-26:
Is it necessary to have CTRL+C as switch to interactive mode? I feel from the beginning that that's a bug but it looks like someone intentionally had implement it.

saqibkh, 2021-02-18:
Is there a spike parameter that can cause "-d" and "not-d" options to behave similarly? i.e. to disable instruction cache?
I'm looking at instruction trace and there is a discrepancy in the register values between the debug and not-debug version that I'd like to resolve.

Thanks