riscv-software-src/riscv-isa-sim #245: Read Retired Instruction CSR Register in Spike

akashgj, 2018-10-05:
I used the following code to get the value of the 'Retired Instruction Register'.

#include <stdio.h>

#define read_csr(reg) ({ unsigned long __tmp;asm volatile ("csrr %0, " #reg : "=r"(__tmp));__tmp; })
#define CSR_INSTRET 0xc02

int main(void) {
long numInstructions;
numInstructions=read_csr(instret);
printf(" %ld\n", numInstructions);
return 0;
}

I cross compiled the code and executed using RISC-V Linux using spike simulator. I am getting values in the range of billions. I am confused whether the program is working correctly?

aswaterman, 2018-10-05:
Are the deltas in the range of billions? The ISA says nothing about the initial value.

akashgj, 2018-10-08:
I just want to know the approximate number of instructions executed by another sample program. So I just take the difference of Retired Instruction Register values before and after the execution of that sample program. The difference in the value of Retired Instruction Register is in range of billions. 

For getting the value of Retired Instruction Register I uses the above code.