riscv-software-src/riscv-isa-sim #248: terminate called after throwing an instance of 'trap_store_access_fault'

IC-Dream, 2018-10-22:
Hi,

when I use "spike -l --isa=RV32IMAFDC testMUL.elf >& out.log", I got the error: terminate called after throwing an instance of 'trap_store_access_fault', I have no idea about this, any help will be appreciated. 

The gcc commad:
[image]

The .ld files:
/* not needed, but we need separate linker scripts anyway */
OUTPUT_ARCH(riscv)

/* required to correctly link newlib */
GROUP( -lc -lgloss -lgcc -lsupc++ )

SEARCH_DIR(.)
__DYNAMIC  =  0;

MEMORY
{
    instrram    : ORIGIN = 0x00000000, LENGTH = 0x8000
    dataram     : ORIGIN = 0x00100000, LENGTH = 0x6000
    stack       : ORIGIN = 0x00106000, LENGTH = 0x2000
}

/* Stack information variables */
_min_stack      = 0x1000;   /* 4K - minimum stack space to reserve */
_stack_len     = LENGTH(stack);
_stack_start   = ORIGIN(stack) + LENGTH(stack);

/* We have to align each sector to word boundaries as our current s19->slm
 * conversion scripts are not able to handle non-word aligned sections. */

SECTIONS
{
    .vectors :
    {
        . = ALIGN(4);
        KEEP(*(.vectors))
    } > instrram

    .text : {
        . = ALIGN(4);
        _stext = .;
        *(.text)
        _etext  =  .;
        __CTOR_LIST__ = .;
        LONG((__CTOR_END__ - __CTOR_LIST__) / 4 - 2)
        *(.ctors)
        LONG(0)
        __CTOR_END__ = .;
        __DTOR_LIST__ = .;
        LONG((__DTOR_END__ - __DTOR_LIST__) / 4 - 2)
        *(.dtors)
        LONG(0)
        __DTOR_END__ = .;
        *(.lit)
        *(.shdata)
        _endtext = .;
    }  > instrram

    /*--------------------------------------------------------------------*/
    /* Global constructor/destructor segement                             */
    /*--------------------------------------------------------------------*/

    .preinit_array     :
    {
      PROVIDE_HIDDEN (__preinit_array_start = .);
      KEEP (*(.preinit_array))
      PROVIDE_HIDDEN (__preinit_array_end = .);
    } > dataram

    .init_array     :
    {
      PROVIDE_HIDDEN (__init_array_start = .);
      KEEP (*(SORT(.init_array.*)))
      KEEP (*(.init_array ))
      PROVIDE_HIDDEN (__init_array_end = .);
    } > dataram

    .fini_array     :
    {
      PROVIDE_HIDDEN (__fini_array_start = .);
      KEEP (*(SORT(.fini_array.*)))
      KEEP (*(.fini_array ))
      PROVIDE_HIDDEN (__fini_array_end = .);
    } > dataram

    .rodata : {
        . = ALIGN(4);
        *(.rodata);
        *(.rodata.*)
    } > dataram

    .shbss :
    {
        . = ALIGN(4);
        *(.shbss)
    } > dataram

    .data : {
        . = ALIGN(4);
        sdata  =  .;
        _sdata  =  .;
        *(.data);
        *(.data.*)
        edata  =  .;
        _edata  =  .;
    } > dataram

    .bss :
    {
        . = ALIGN(4);
        _bss_start = .;
        *(.bss)
        *(.bss.*)
        *(.sbss)
        *(.sbss.*)
        *(COMMON)
        _bss_end = .;
    } > dataram

    /* ensure there is enough room for stack */
    .stack (NOLOAD): {
        . = ALIGN(4);
        . = . + _min_stack ;
        . = ALIGN(4);
        stack = . ;
        _stack = . ;
    } > stack

    .stab  0 (NOLOAD) :
    {
        [ .stab ]
    }

    .stabstr  0 (NOLOAD) :
    {
        [ .stabstr ]
    }

    .bss :
    {
        . = ALIGN(4);
        _end = .;
    } > dataram
}

Thanks
Dream

IC-Dream, 2018-10-23:
Hi, @ethanhz,

here is the disassembly:
[image]
I think that the "sw " of line 12 ~ 26 might introduce this error.

Thanks
Dream

ethanhz, 2018-10-23:
Actually, line#8 will already (potentially) give this error.
Can you modify your linker so the code starts from 0x80000_0000?
This is a commonly used starting code address since some lower addresses in Spike (and other available implementions I have seen) are reserved for specific purposes.
Hope this helps.

IC-Dream, 2018-10-23:
Thanks a lot, Yea, i noticed that the linker is always starts from 0x8000_0000 in riscv-test(https://github.com/riscv/riscv-tests), and I try to modify the code start address to 0x8000_0000, there is 
some build error. actually, i am evaluating the pulpino core(https://github.com/pulp-platform/pulpino).

IC-Dream, 2018-10-23:
so, could you please show me how to allocate the address space in spike ? or any reference for spike ? thanks a lot.

ethanhz, 2018-10-23:
spike ..... **-m0xFFFF00000000:0x1000**  <ELF_FILE>
This will add 0x1000 bytes of region starting at addr: 0xFFFF00000000.

But that's not what you want for this issue. For this issue, you need to try changing your linker.

Have you tried changing the below linker lines:
    _instrram    : ORIGIN = 0x00000000, LENGTH = 0x8000
    dataram     : ORIGIN = 0x00100000, LENGTH = 0x6000
    stack       : ORIGIN = 0x00106000, LENGTH = 0x2000_

Regards,
Rohit

ethanhz, 2018-10-23:
spike ..... **-m0xFFFF00000000:0x1000**  <ELF_FILE>
This will add 0x1000 bytes of region starting at addr: 0xFFFF00000000.

But that's not what you want for this issue. For this issue, you need to try changing your linker.

Have you tried changing the below linker lines:
    _instrram    : ORIGIN = 0x00000000, LENGTH = 0x8000
    dataram     : ORIGIN = 0x00100000, LENGTH = 0x6000
    stack       : ORIGIN = 0x00106000, LENGTH = 0x2000_

IC-Dream, 2018-10-23:
Yea, i have tried modified these lines you pointed out like this, but i would get some build errors.
instrram : ORIGIN = 0x80000000, LENGTH = 0x8000
dataram : ORIGIN = 0x80100000, LENGTH = 0x6000
stack : ORIGIN = 0x80106000, LENGTH = 0x2000

scottj97, 2023-01-22:
There is also the debug ROM at 0x0 (see `DEBUG_START` in `riscv/decode.h`) and I suspect that's where the conflict lies.

When I run your ELF on a modern Spike, I get:
```
Access exception occurred while loading payload riscv_arithmetic_basic_test_1.elf:
Memory address 0x80 is invalid
```

abdullahyildiz, 2023-01-23:
@scottj97,

Thanks for the information. I have been working on [v.1.1.0](https://github.com/riscv-software-src/riscv-isa-sim/tree/v1.1.0) and tried to reconfigure the `DEBUG_START` and `DEBUG_END` accordingly; however, there must be something else I have been missing so far.

[hello-world.objdump.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/10478034/hello-world.objdump.txt)
[hello-world.elf.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/10478033/hello-world.elf.txt)

ProxyPlayerHD, 2023-10-03:
I'm also getting this error but cannot figure out why. i changed my linker script to place code at 0x8000000 as was suggested on here, and i still don't know why it doesn't work...

command: `./spike -l -m0x80000000:0x100000 --pc=0x80000000 test.elf --log=output.txt`

i run it as a RV32IMAC CPU (Machine Mode only), while the program is compiled for RV32IM (it's just a blank main() function so there should be no incompatibilities).

program disassembly:

```test.elf:     file format elf32-littleriscv

Disassembly of section .init:

80000000 <_start>:
80000000:	00002197          	auipc	x3,0x2
80000004:	86818193          	add	x3,x3,-1944 # 80001868 <__global_pointer$>
80000008:	00080117          	auipc	x2,0x80
8000000c:	bf810113          	add	x2,x2,-1032 # 8007fc00 <__stack_top>
80000010:	00010433          	add	x8,x2,x0
80000014:	008000ef          	jal	8000001c <main>
80000018:	00000067          	jr	x0 # 0 <_start-0x80000000>

Disassembly of section .text:

8000001c <main>:
8000001c:	ff010113          	add	x2,x2,-16
80000020:	00812623          	sw	x8,12(x2)  # (this is the only store instruction in the entire program, but it writes to 0x8007FC0C which is well within the memory region)
80000024:	01010413          	add	x8,x2,16
80000028:	00000793          	li	x15,0
8000002c:	00078513          	mv	x10,x15
80000030:	00c12403          	lw	x8,12(x2)
80000034:	01010113          	add	x2,x2,16
80000038:	00008067          	ret

Disassembly of section .eh_frame:

8000003c <__BSS_END__-0x102c>:
8000003c:	0014                	.insn	2, 0x0014
8000003e:	0000                	.insn	2, 0x
80000040:	0000                	.insn	2, 0x
80000042:	0000                	.insn	2, 0x
80000044:	00527a03          	.insn	4, 0x00527a03
80000048:	7c01                	.insn	2, 0x7c01
8000004a:	0101                	.insn	2, 0x0101
8000004c:	07020d1b          	.insn	4, 0x07020d1b
80000050:	0001                	.insn	2, 0x0001
80000052:	0000                	.insn	2, 0x
80000054:	0010                	.insn	2, 0x0010
80000056:	0000                	.insn	2, 0x
80000058:	001c                	.insn	2, 0x001c
8000005a:	0000                	.insn	2, 0x
8000005c:	ffa4                	.insn	2, 0xffa4
8000005e:	ffff                	.insn	2, 0xffff
80000060:	001c                	.insn	2, 0x001c
80000062:	0000                	.insn	2, 0x
80000064:	0000                	.insn	2, 0x
	...

Disassembly of section .comment:

00000000 <.comment>:
   0:	3a434347          	.insn	4, 0x3a434347
   4:	2820                	.insn	2, 0x2820
   6:	29554e47          	.insn	4, 0x29554e47
   a:	3120                	.insn	2, 0x3120
   c:	2e32                	.insn	2, 0x2e32
   e:	2e32                	.insn	2, 0x2e32
  10:	0030                	.insn	2, 0x0030

Disassembly of section .riscv.attributes:

00000000 <.riscv.attributes>:
   0:	2941                	.insn	2, 0x2941
   2:	0000                	.insn	2, 0x
   4:	7200                	.insn	2, 0x7200
   6:	7369                	.insn	2, 0x7369
   8:	01007663          	bgeu	x0,x16,14 <_start-0x7fffffec>
   c:	001f 0000 1004      	.insn	6, 0x10040000001f
  12:	7205                	.insn	2, 0x7205
  14:	3376                	.insn	2, 0x3376
  16:	6932                	.insn	2, 0x6932
  18:	7032                	.insn	2, 0x7032
  1a:	5f31                	.insn	2, 0x5f31
  1c:	326d                	.insn	2, 0x326d
  1e:	3070                	.insn	2, 0x3070
  20:	7a5f 6d6d 6c75      	.insn	6, 0x6c756d6d7a5f
  26:	7031                	.insn	2, 0x7031
  28:	0030                	.insn	2, 0x0030
```

so what could i be missing here?

ProxyPlayerHD, 2023-10-06:
the version i built is the current stable release 1.1.0.... though true it is quite old
i'll download the repo itself and build that and see what it does. i'll report back